f4exb/sdrangel #595: QT virtual keyboard

hiperiondev, 2020-08-11:
I'm using sdrangel on a touch screen device.
The problem is that is not possible to change frequency without a physical keyboard .
In others programs (like kterm) I use QT_IM_MODULE=qtvirtualkeyboard to start on screen keyboard but in sdrangel doesn't work

Vort, 2020-08-12:
I was able to make keyboard appear, but:
1. It works only for main window and not for popup windows (clicks on keys do nothing there).
2. It not appears when I click on frequency controls (maybe they are custom and have no support for virtual keyboard).

Vort, 2020-08-12:
Looks like I was able to make it work.
@hiperiondev, do you know how to test patches? Can you look if this change is ok: https://github.com/Vort/sdrangel/commit/73b221f05515ca8257db854bae974bc8231d312d ?

hiperiondev, 2020-08-12:
I'll try to recompile sdrangel this afternoon

hiperiondev, 2020-08-12:
Doesn't work and can't change dial with physical keyboard now

hiperiondev, 2020-08-12:
No start virtual keyboard on dial.

Vort, 2020-08-12:
> No start virtual keyboard on dial.

Do it shows with another UI controls?
Like server address in KiwiSDR input plugin.

hiperiondev, 2020-08-12:
My bad!!! Sorry.
Yes, it's working on dial and others UI controls

closed 2020-08-12 by hiperiondev

Vort, 2020-08-13:
@f4exb did not reviewed the fix yet, so no need to close report yet.

Vort, 2020-08-13:
I made pull request for this fix: #598.

PR #598 (2020-08-13): ValueDial(Z): add virtual keyboard support
Fixes #595.