f4exb/sdrangel #596: I/Q audio sink with SoapyAudio?

hiperiondev, 2020-08-12:
Is it possible to send the signal in transmission through the audio card to drive a cheap sdr transmitter? maybe through SoapyAudio?

Vort, 2020-08-13:
Looks like transmission support in SoapyAudio is broken :(
_(but, maybe, I am trying to use it in a wrong way, do not know)_
<details>
<summary>Screenshot of SoapySDRUtil output</summary>

[image]

</details>

f4exb, 2020-08-13:
Soapy being not a that great option this ticket was open for direct Audio support: https://github.com/f4exb/sdrangel/issues/496
So it is in the plans...

f4exb, 2020-09-18:
To be superseded by #496

closed 2020-09-18 by f4exb