f4exb/sdrangel #600: AirspyHF does not seem to work properly in Ubuntu 20.04

f4exb, 2020-08-13:
After upgrade to 20.04 the AirspyHF is still recognized and opens but yields no I/Q data. Not sure this is not a side effect of something else or something in my own environment (I had to struggle a bit to make 20.04 functional) but deserves to be looked at.

f4exb, 2020-08-14:
It turns out that after call to:
```cpp
airspyhf_error rc = (airspyhf_error) airspyhf_start(m_dev, rx_callback, 0);
```
`rx_callback` is never called

f4exb, 2020-08-15:
Airspy has the same structure and it works so this remains a mystery.

srcejon, 2020-08-25:
I've just installed Ubuntu 20.04.1 LTS (as a VM under a Windows host) and built sdrangel according to https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux and airspyhf is working for me.

f4exb, 2020-08-25:
Thanks for the report. It could be specific to my configuration.

zipzoop11, 2020-09-04:
Built sdrangel according to instructions  https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux
Running on a Lenovo ThinkPad T590 and am experiencing the same issue as originally described.
Have confirmed software works as expected with other SDR hardware.
Issue also seems to occur on Debian Buster.

f4exb, 2020-10-06:
At some point libusb was downgraded to 1.0.21 in the windows libraries: https://github.com/f4exb/sdrangel-windows-libraries/commit/07e9c2ee5fe748decbd3679a08d43da53b7e239f I don't remember the exact reason but it may be worth trying with 1.0.21. In the Ubutnu 20.04 distribution the libusb library has been upgraded to 1.0.23

xplanefan, 2020-10-10:
Confirmed that reverting to libusb v1.0.21 does indeed resolve the issues on Debian 10.6 Buster. 

Still need to try latest 1.0.23 HEAD.

f4exb, 2020-10-14:
:tada: Breaking news: I have upgraded the stock 5.4.0 kernel of Ubuntu 20.04 to latest 5.8.14 using the `.deb` available from PPA as documented here https://itsfoss.com/upgrade-linux-kernel-ubuntu/ and now my Airspy HF wants to throw I/Q samples again.

f4exb, 2020-10-15:
Documented in the Wiki: https://github.com/f4exb/sdrangel/wiki/Sample-source-plugins-(Rx-devices)

closed 2020-10-15 by f4exb

alphafox02, 2022-01-21:
Would it be possible to revisit this? I finally got ahold of the Airspy HF+ Discover and can confirm that while SDRAngel doesn’t show any output or appear to work with it - the same exact system is capable of having SDR++, GQRX, and SigDigger so far working with the Airspy HF+ Discovery with absolutely no change in software, libusb, or kernel. It’s only SDRAngel installed from the deb package. 

I will happily provide more info. I’m trying to avoid system changes when all other tested applications so far work with the HF+. 

Thank you.

alphafox02, 2022-01-22:
I tried for a couple hours, along with someone else, to make some changes. No luck so far. The Airspy dev suggested initially this 

static int __cdecl rx_callback(airspyhf_transfer_t* transfer);

however I tried the GCC equivalent which has cdecl after the functions, which did compiled, but provided no change. The reason this was suggested I think was because of both your mention above about no rx callback and then the hint about the kernel. I think maybe you can have a working solution with the 5.4 kernel. Here’s more info.

“maybe it's the ABI of the compiler used with the kernel

the ABI is only stable when using explicit calling conventions

it is the case with the library (check ADDCALL define)”