f4exb/sdrangel #602: hamlib rigctl support?

hiperiondev, 2020-08-15:
Programs like Gpredict can correct on the fly frequency for Doppler effect on satellite communications. Usually use rigctl for this.
There is some support for this in sdgrangel?

f4exb, 2020-08-16:
No this is not the path taken in SDRangel. SDRangel can be fully controlled with its http REST API and this is the way to go to implement such things like Doppler shift control. In order to do this you have to write some intermediary software that will make the joint between data from Gpredict and SDRangel. This can be done in Python for example. There are examples in the `scriptsapi` folder and also in `swagger/sdrangel/examples`.

srcejon, 2020-09-08:
@hiperiondev, I've added a plugin that should support rigctrl in pull request #627 If you can try it out, it would be good to get some feedback from you.

f4exb, 2020-09-08:
This is a more general consideration but I start to like the idea of a "functional" plugin that this development makes emerging. I have only looked at the UI interface for the moment and I see that it is still pretty much embedded in the core (main) functions technically attached to the "Preferences" menu. 

Now what if this plugin would be able to generate several independent instances of itself like the channel plugins do? Then this would make it possible to control more than one device and one channel each instance opening its own port to receive commands. This is more in line with the spirit of SDRangel since its inception. And when I look at other items in the preferences menu the "LimeRFE control" is a good candidate also for a "functional" plugin, "AMBE" maybe also. It also makes it easier to manage dependencies if you just don't care about these functions and do not want to deal with SerialDV or LimeSuite dependencies. 

Similarly to what exists for channels then a canvas must be added to lay out the UI of these plugins and this adds even more stuff in the screen space which is pretty crowded already. However canvas can be stacked and this should not be a consideration that would prevent a development that could bring significant added value to the software.

For now I will keep it more or less like this as I need to put more thinking into this idea but some reorganization may take place in the future.

f4exb, 2020-09-08:
Two issues have to be addressed in the mid term:
  - `sdrbase/settings` should have no mention of `PluginManager`. The save/load of the RigCtrl plugin preferences should be handled in a different way
  - there is (or there should be) the possibility to retrieve the own instance API host address and port. These should be used with no possibility for the user to enter these parameters. This plugin should not be able to control other instances of SDRangel.

srcejon, 2020-09-09:
Yes, I agree the API address shouldn't be a user setting. I just didn't want to make too many changes to the existing internal API without getting some feedback from you. If you have a suggestion for that and how the settings should be stored, I'm happy to redo the plugin.

f4exb, 2020-09-10:
I have made a few changes after merge that are now on the dev branch. Mostly removing some dependencies on `QtWidget` and `QMainWindow`. The `MainWindow` that inherits from `QMainWindow` is a singleton class so it can be accessed without explicitly carrying its pointer over and over. This way one can get access to the API host and port anywhere and this is presently done in `RigCtrlPlugin::createTopLevelGUI()`. At the moment this will work only for the GUI variant because there is no provision for creating an instance of the server variant from the plugin interface. I don't think this is an issue because it will not be used in the server variant anyway.

A much deeper redesign has to be carried out in order to accept these "feature" plugins as first class citizens similarly to "channel" plugins which will solve a lot of dependency issues and integrate the new code in a better way. For now it is OK to have it a bit as a stowaway but to accept more of these kind of plugins some architectural changes have to be made.

The idea of integrating specific features as plugins is a very good idea and I want to thank you for emitting the spark. Now with my knowledge of this project architecture as a whole it is probably my job to do the final integration steps.

srcejon, 2020-09-10:
Sounds good. Thanks Edouard.

hiperiondev, 2020-09-10:
> @hiperiondev, I've added a plugin that should support rigctrl in pull request #627 If you can try it out, it would be good to get some feedback from you.

Good news!
I'll try your plugin this weekend.

f4exb, 2020-09-13:
Implemented in v4.15.5

closed 2020-09-13 by f4exb