f4exb/sdrangel #603: [question] would a CW demodulator be possible?

febs, 2020-08-15:
I think I stumbled upon some morse code transmissions while listening with SDRangel (it was around 7MHz) and I though it would be super cool if the software could automatically translate that into text for me. 

Thanks,

f4exb, 2020-08-16:
I have though about it but in order to do this properly and go beyond the traditional threshold approach that does not work very well unless you have a calibrated strong signal to process you have to resort to machine learning and this is not a small undertaking. I think I have come across some works that have been done in this area but I don't have the pointers right now (to be added later). One can also dig into speech recognition which could be used in a similar approach.

f4exb, 2020-08-17:
The works I came across was Mauri Niininen's (AG1LE). I don't have the original pointers but there seems to be more recent ones:
  - [Article in Hackaday](https://hackaday.com/2020/01/27/machine-learning-system-uses-images-to-teach-itself-morse-code/)
  - [AG1LE blog post](http://ag1le.blogspot.com/2019/02/training-computer-to-listen-and-decode.html)

hiperiondev, 2020-10-15:
Maybe this can be useful

http://ag1le.blogspot.com/2013/09/new-morse-decoder-part-1.html

f4exb, 2020-12-14:
The proposed technique is derived from Handwritten Text Recognition (HWT). In the recent years Transformers have superseded LSTM. 
  - https://blog.parashift.io/blog/handwritten-text-recognition (overview of HWT recognition techniques including Transformers)
  - https://arxiv.org/abs/2005.13044 (included link)

f4exb, 2020-12-23:
I have explored a bit more the subject and Mauri's AG1LE work. Actually I found his older work: https://github.com/ag1le/RNN-Morse quite interesting. Although less elaborate than the more recent developments it seems to take into account the "music" of the Morse code that is the "dit"` and "dah" lengths and the various forms of silence. Morse is not quite like handwriting so I am not sure the best approach is to reapply HCR methods directly on it. Any proficient human CW "decoder" will tell you that the "music of the code" is very important. So not taking this into account at the very first step might be detrimental to the rest of the chain. The "dit", "dah" and various silence separators input could then be taken as stimuli of the next network. As a very first step these stimuli lines could be used to make a powerful "intelligent" CW filter.

f4exb, 2020-12-30:
After 10 days of intensive work there is no progress. Hence this issue is closed

f4exb, 2021-01-15:
Some food for thought here: https://github.com/f4exb/morseangel This is only in Jupyter Notebooks so at research stage. The `RNN-Morse-chars_dual` notebook looks promising but I still have the lack of knowledge to properly extract the final data.

f4exb, 2021-01-29:
Eventually something that seems to work fairly well: https://github.com/f4exb/morseangel/blob/main/notebooks/RNN-Morse-chars_single-ord36e96.ipynb Training needs to be done progressively and with a rather low learning rate from the start. This model is fairly hard to train because its sweet spot seems to be very narrow. The effective number of samples per dit is increased to 7.69 (decimation 96) which allows some more tolerance.

f4exb, 2021-02-05:
[MorseAngel](https://github.com/f4exb/morseangel ) is now an autonomous audio based application. It uses PyQt5 for the GUI and has the same theme as SDRangel. I think there is no point in integrating it with SDRangel. It is very easy to pipe in the audio coming out of SDRangel at best a mention can be made in the Wiki. At the moment this is not working satisfactorily but can show some intermittent successful decodes. Development will continue (contributors welcome!) in the MorseAngel project.

closed 2021-02-05 by f4exb

mfalkvidd, 2023-05-07:
New link (I guess the page was renamed): https://github.com/f4exb/sdrangel/wiki/Decoding-Morse-code-from-audio