f4exb/sdrangel #604: Fix libiio channel handling for IQ sampling with plutosdr (specifically fw 0.32)

ubfx, 2020-08-19:
Hi,

as others have pointed out, there are problems with plutosdr fw 0.32 and sdrangel. I have not tested plutosdr with older firmware but seem to have found the cause of the problem.

Example: I'm injecting a 433.0 MHz sine into the RX port of the pluto. This is what I see with the current release of SDRAngel:
[image]

There's obviously some mirroring going on which should not be the case if the IQ samples were processed correctly. Also, it's not a hardware issue because in MATLAB/Simulink, I can correctly see the baseband.

I've checked in the SDRAngel source code and the problem seems to be that only one RX channel ("voltage0") is opened via libiio. This is the I channel. The Q channel is "voltage1". This leads to a situation where only I samples are in the sample buffer but they are interpreted as the real and imaginary part of IQ samples. This leads to the mirroring effect that can be seen in my above screenshot (and also to aliasing).

Actually, I'm kind of surprised that this seemed to work with older firmware versions, maybe in older firmwares, both I and Q channels were opened automatically when the I channel was opened for RX. I think this should be investigated further. However, since MATLAB/Simulink seems to be able to handle all fw versions correctly, there must be a way to make it work across all the firmware versions.

I've implemented a fix (**only tested with fw 0.32**) for reference: https://github.com/ubfx/sdrangel/tree/plutosdr_iq_sampling
Screenshot of the same signal with the fixed version:

[image]

Related:
- https://github.com/f4exb/sdrangel/issues/589
- https://github.com/analogdevicesinc/plutosdr-fw/issues/53

ubfx, 2020-08-19:
I just tested the fix with fw 0.31 and it works. Cleaned it up a bit and created a PR to dev for further testind and evaluation:

https://github.com/f4exb/sdrangel/pull/605

f4exb, 2020-08-19:
Many thanks! I think this makes sense and a default opening of both channels was possibly the case with FW < 0.32 this could even be a bug that has been corrected because I suppose the authors want to make it possible to open I and Q channels individually. In some applications this may make sense.

f4exb, 2020-08-19:
A comment in the old PlutoSDRThread source said "I and Q samples are processed one after the other" this could be a hint. Previously it was possible to open the I channel only ("voltage 0") and then I and Q samples were available in sequence but this method was deprecated somehow (didn't check if it was mentioned in libiio doc or release notes). Now it is only possible to have I and Q on independent channels.

f4exb, 2020-08-19:
I confirm this fixes the issue and is compatible with FW 0.31

f4exb, 2020-08-19:
Tx side was already working with 2 channels for I and Q so no changes are needed there.

ubfx, 2020-08-19:
Thanks for looking into it so quickly! I'm glad it also fixes the audio issue.

closed 2020-08-19 by ubfx

roelandjansen, 2020-08-20:
Thanks for the swift stuff! Wanted to mention this before but at work I cannot do anything with github if the repo is not theirs.

Short q though:

I swicthed to the dev branch:

commit 0c04560b3c7b0a52265b1e17f807af538e3ef4a9 (HEAD -> dev, origin/dev)
Merge: 596d2ef14 c4ccb59e9
Author: f4exb <f4exb06@gmail.com>
Date:   Wed Aug 19 13:13:45 2020 +0200
    Merge pull request #605 from ubfx/plutosdr_iq_sampling
    Fix libiio channel handling for IQ sampling with plutosdr

but it still stutters w/ v0.32.

Wondering if I forgot something.

roelandjansen, 2020-08-20:
thanks for the swift code changes! One laptop fails; the other works fine. Probably my fault.

ubfx, 2020-08-20:
That's interesting, did it ever work on that one laptop that's not working? Are you using low sample rates in case the USB can't keep up?