f4exb/sdrangel #606: Send i/q signals to local process

hiperiondev, 2020-08-20:
Is it possible to send the modulated i/q signals of a sink?
I think that a local output would be correct but it is not clear in the documentation how to indicate the process or pipe and what data would be received

hiperiondev, 2020-08-20:
This question is associated with the need to output these signals by audio while waiting for the full implementation to take place (perhaps with gnu-radio for now).
In my case I have a sdr-rtl for reception and two sound cards, one will be used for audio and microphone and the other for the output to the sdr

f4exb, 2020-08-21:
As the name implies "local" stays local to the SDRangel instance and these plugins can be used for transfer between device sets (i.e. tabs in the GUI).
  - LocalOutput in one tab -> LocalSource in another tab (for Tx)
  - LocalSink in one tab -> LocalInput in another tab (for Rx)

f4exb, 2020-11-04:
Question answered

closed 2020-11-04 by f4exb