f4exb/sdrangel #608: GUI glitch with v￼4.15.0+/v5.9.0￼+

dmtrbm, 2020-08-23:
When collapsing device pane by clicking the triangle - it disappears completely, making it impossible to expand again. 
[image]
Versions v4.14.20/v5.8.5 work as expected. Linux, qt-5.15.0.

Thank you.

f4exb, 2020-08-24:
Nothing to do with Qt or glibc versions it happens with Qt 5.12.8. I suspect a side effect of commit d80d050992e87e30f5a4ad560a5d06a6b921442e

f4exb, 2020-08-24:
A quick check confirms that reverting this commit fixes the issue. This does not change the fact that the original change was supposed to make things better when a dialog is opened from this widget. The RollupWidget base design is bad. This can be covered in another more specific issue.

commit 0a562301 (2020-08-25): RollupWidget: fixed collapse section header not showing up. Fixes #608
commit 603ff292 (2020-08-25): RollupWidget: fixed collapse section header not showing up. Fixes #608
closed 2020-08-25 by f4exb