f4exb/sdrangel #609: Spectrum/waterfall pane proportions no longer saved with presets.

dmtrbm, 2020-08-23:
Prior to v4.14.16/v5.6.0, each preset would remember spectrum/waterfall height.
This no longer works and presets revert to default proportions on restart.
Thanks.

f4exb, 2020-08-26:
Sorry this is not what I see. Can you detail exactly what you are doing?

dmtrbm, 2020-08-28:
Start with empty config, resize spectrum to its minimum height, save as preset A.
Resize spectrum to its maximum height, save as preset B.
Load preset A, spectrum remains at its maximum height (it should revert back to minimum).
Restart, both presets have default proportions.

f4exb, 2020-08-29:
By "resize spectrum" do you mean set the ratio between spectrum graph and waterfall? This very specific part of the setup does not seem to be serialized and/or deserialized.

f4exb, 2020-08-29:
OK I see. There is indeed an issue because the waterfall share factor that is controlled by GLSpectrum is not saved in the settings that are now completely managed by GLSpectrumGUI. GLSpectrum has to report it to GLSpectrumGUI for it to update the settings. This is a fairly simple fix.

closed 2020-08-29 by f4exb

dmtrbm, 2020-11-14:
Looks like this issue is back in 6.0.0.

commit 1429e3ac (2020-08-29): GLSpectrum: fixed settings and waterfall share serialization. Fixes #609
commit 97efda73 (2020-08-29): GLSpectrum: fixed settings and waterfall share serialization. Fixes #609