f4exb/sdrangel #614: SDRIQ files with a playtime (record length) of less than 1 second can not be played

bkarl, 2020-08-24:
If an SDRIQ file has a playtime (record length) of less than one second (eg samplerate 100 kHz and 50 kS in the file) the file will not be played by the "FileInput" device nor by the "File Channel Source". This happens because the integer m_recordLength (playtime in seconds) gets rounded towards zero and the file stream is closed right after opening

Because the CRC is still valid the software either gets in an invalid state and does not stop playing or does not play the file at all (Warning: "FileInput::start: file not open. not starting").

The corresponding code is at 
https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/fileinput/fileinput.cpp#L155 and 
https://github.com/f4exb/sdrangel/blob/master/plugins/channeltx/filesource/filesourcesource.cpp#L227

Are patches welcome for this or is it expected behavior? Thank you.

f4exb, 2020-08-24:
I doubt a file of less than a second length will produce anything useful do you have a concrete use case for this?

bkarl, 2020-08-25:
For some communication systems short data bursts are used, eg
- a TETRA frame has 56.67 ms length
- Wireless MBus (EN 13757-4) also uses short data bursts

In my case I got Matlab generated data bursts for a proprietary digital communication system that I want to transmit using LimeSDR. My current workaround is to pad the data with zeros until a second of playtime is reached. 

I can totally understand that this is not a standard ham radio use case but it would be nice to get a warning or error message at least or document the behavior in the file format spec.

f4exb, 2020-08-25:
Well I suppose you are going to play the sequence in a loop aren't you? So my concern is the loop turnaround time and the possible glitches in the transient period creating more issues than it is supposed to solve. However there is no limitation in principle to use sub-second sequences.

bkarl, 2020-08-25:
I get your point. For now I'm not playing the bursts in a loop but you are right, this might be an issue. I will implement the sub-second feature, check if looping is an issue and come back with a PR or comment here. Thanks for your time!

f4exb, 2020-08-25:
Well I also started on my side to check for possible side effects. What I did is transformed `m_recordLength` to ` m_recordLengthMuSec` changing seconds value to microsecond value and adapting the various calculations. This seems to be working fine.

f4exb, 2020-08-25:
I have added milliseconds to the total time display also which is more consistent with the rest of the time displays:
[image]

f4exb, 2020-08-27:
Implemented in v4.15.3 and v5.9.3

closed 2020-08-27 by f4exb

bkarl, 2020-08-27:
Awesome! Thank you for this great project.