f4exb/sdrangel #615: Closing channel analyzer causes segfault

srcejon, 2020-08-26:
With sdrangel-286-master.tar.gz 4.15.2 release on Ubuntu 20.04.1 LTS Qt 5.12.8 (and also version I've compiled from source), closing the channel analyzer causes a segfault. It seems very repeatable. To reproduce:

Run sdrangel
Add Channel Analzyer
Close Channel Analzyer
Segfault

Last few messages from console:

2020-08-26 02:16:38.974 (D) DSPDeviceSourceEngine::removeSink:  ChannelAnalyzer
2020-08-26 02:16:38.974 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2020-08-26 02:16:38.975 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT fwd size: 8192
2020-08-26 02:16:38.975 (D) FFTFactory::releaseEngine: engineSequence: 1 FFT fwd size: 8192
2020-08-26 02:16:38.975 (D) FFTFactory::releaseEngine: engineSequence: 0 FFT inv size: 8192
2020-08-26 02:16:38.975 (D) FFTFactory::releaseEngine: engineSequence: 2 FFT fwd size: 1024
2020-08-26 02:16:38.975 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440046e0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
2020-08-26 02:16:38.976 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440046e0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
2020-08-26 02:16:38.976 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440046e0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
2020-08-26 02:16:38.976 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440046e0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
2020-08-26 02:16:38.980 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440112d0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
2020-08-26 02:16:38.980 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440112d0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
2020-08-26 02:16:38.980 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440112d0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
2020-08-26 02:16:38.980 (W) QOpenGLTexturePrivate::destroy() called but texture context 0x7f86440112d0 is not shared with current context 0x7f864400b5c0.
Texture has not been destroyed
Segmentation fault (core dumped)

f4exb, 2020-08-26:
I noticed that and have not found the cause of the issue. The core dump is pretty slim.

f4exb, 2020-08-26:
Interestingly this does not happen in Ubuntu 18.04 with Qt 5.9.5 ...

f4exb, 2020-10-21:
Does not happen when compiled with Qt 5.14.2 (Ubuntu 20.10). In 20.04 Qt 5.12.8 is used so at least with this version the issue appears. 

A drastic solution would be to upgrade to Ubuntu 20.10 which has also the advantage of upgrading to kernel 5.8 series fixing the issue with Airspy HF. Qt 5.14.2 has a bug at compile time though but that can be worked around with the `-fpermissive` flag.

f4exb, 2020-11-06:
Documented in the [Wiki](https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux)

closed 2020-11-06 by f4exb

f4exb, 2021-02-15:
This is a completely different issue. In fact you attempt to delete channel with index 1 twice. However this should return 404 because the channel does not exist anymore at the second call so there could be a bug in the HTTP request handling. 

In any case when using the API you have to keep track of your actions. This is a stateless API. To achieve what you are trying to do you should apply these two requests in sequence:

```
(DELETE) /sdrangel/deviceset/0/channel/1
(DELETE) /sdrangel/deviceset/0/channel/0
```

Valstee, 2021-02-15:
Thanks for the quick answer 👍 
Indeed, there is an error in what I shared, I had created 3 channels, So : `(DELETE) /sdrangel/deviceset/0/channel/1` should have worked.

So it works with these two requests in sequence :
```
(DELETE) /sdrangel/deviceset/0/channel/1
(DELETE) /sdrangel/deviceset/0/channel/0
```

But not with these :
```
(DELETE) /sdrangel/deviceset/0/channel/0
(DELETE) /sdrangel/deviceset/0/channel/0
```

f4exb, 2021-02-17:
Issue #776 opened.

alphafox02, 2022-03-18:
In reference to the original ticket problem - is there any recent information that could correct this problem with the Qt version in 20.04? I’m looking around and finding, at first glance, possibly similar problems with potential workarounds.

https://bugreports.qt.io/plugins/servlet/mobile#issue/AUTOSUITE-220

alphafox02, 2022-03-18:
“Workaround
Implement closeEvent() somewhere and cleanup opengl resources wile the event loop is still running. This gives you the possibility to call makeCurrent() and will get the textures deleted without warning or errors.”

alphafox02, 2022-05-03:
This is no longer an issue at this point with the latest v7 on 20.04 w/ stock Qt. Closing the analyzer does not cause a seg fault.