f4exb/sdrangel #616: Graphics corruption when run as root

srcejon, 2020-08-26:
When sdrangel 4.15.2 (release sdrangel-286-master.tar.gz or compiled from source) is run as root via sudo on Ubuntu 20.04.1 LTS (in a VMWare VM), the graphics can be corrupted, as in the attached image.

[image]

No problems when run under a user account. Yes, it's possibly not an application that should be run as root (so may be marked as wont fix), but it can be useful when trying to see if USB devices that do not have correct UDEV rules set up for are usable, so I thought I'd mention it.

Vort, 2020-08-26:
Maybe VMware have bad drivers for graphics hardware acceleration.
Can you try to test SDRangel with acceleration disabled in VM settings?
Here is my test result of `sudo sdrangel`:
[image]

f4exb, 2020-08-26:
Also I strongly do not recommend to run as root this should not be necessary and maybe you should check why you resort to run as root.

f4exb, 2020-08-26:
Secondly I had a hard time trying to make the graphics work properly with Ubuntu 20.04 based Docker images so there could be something fishy with it. This happened with both 20.04 and 18.04 hosts so most probably related to the base distribution of the image and not the host. I finally resorted to base my images on Debian Buster (See: https://github.com/f4exb/sdrangel-docker/issues/21)

srcejon, 2020-08-26:
> Maybe VMware have bad drivers for graphics hardware acceleration.
> Can you try to test SDRangel with acceleration disabled in VM settings?

The problem still occurs with acceleration disabled. Note that it displays fine with acceleration enabled, when run from under a normal user account. Only seems to be a problem as root.

> Also I strongly do not recommend to run as root this should not be 
>necessary and maybe you should check why you resort to run as root.

I was just running as root in order to access a USB radio for which the udev rules were not correct. I've now fixed that, so do not have a need to run as root. I just thought I'd mention the problem as I'd seen it (and perhaps it can occur under other conditions). By all means close this if not really worth investigating.

Great s/w BTW 👍

Vort, 2020-08-26:
@srcejon it may be a problem not with SDRangel, but with OS, with VM software, with drivers.
To investigate problem further either you need to remove problem somehow or someone needs to reproduce it on his VM installation.
Another tests can be: running guest OS from the same virtual disk image, but with VirtualBox; installing different OS (Windows, older Linux) on USB stick and trying to boot from it as host and run guest from it.

f4exb, 2020-12-09:
I don't think we can go any further. As already said there are some OS related issues when running in some virtual environments that cannot be related to code issues. Hence closed.

closed 2020-12-09 by f4exb