f4exb/sdrangel #617: USRP radio support via SoapyUHD

srcejon, 2020-08-26:
Ettus USRP radios (such as the B210) can be made to work with SDRangel on Linux via SoapyUHD.

Below, I try to build it in the same way as the other modules on: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

Compile UHD from source, based on: https://files.ettus.com/manual/page_build_guide.html
```
sudo apt-get install libboost-all-dev libusb-1.0-0-dev python-mako doxygen python-docutils cmake build-essential
cd /opt/build
git clone https://github.com/EttusResearch/uhd.git
cd uhd/host
mkdir build
cd build
cmake -DCMAKE_INSTALL_PREFIX=/opt/install/uhd ../
make 
make install
/opt/install/uhd/lib/uhd/utils/uhd_images_downloader.py
# The following aren't required if installed to /
echo /opt/install/uhd/lib | sudo dd of=/etc/ld.so.conf.d/uhd.conf
sudo ldconfig
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/opt/install/uhd/lib/pkgconfig
# Enable USB access from user accounts
cd /opt/install/uhd/lib/uhd/utils
sudo cp uhd-usrp.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules
sudo udevadm trigger
```

Build SoapyUHD:
```
cd /opt/build
git clone https://github.com/pothosware/SoapyUHD.git
cd SoapyUHD
mkdir build; cd build
cmake -DCMAKE_INSTALL_PREFIX=/opt/install/SoapySDR ..
make
make install
```

With that, I have the USRP USB B210 channel 0 working for RX in SDRangel 👍 

Channel 1 fails with:
terminate called after throwing an instance of 'uhd::runtime_error'
  what():  RuntimeError: Invalid recv stream command - stream now on multiple channels in a single streamer will fail to time align.

Most of the device specific settings seem to work. E.g. AGC, bus format, SR etc. One strange thing is when clearing the 'Auto DC corr' checkbox, the 'Global gain' is set to 0, ignoring what value it is set to. It gets the correct value when the checkbox is set though.

srcejon, 2020-08-27:
USRP radios can also work on Windows via Soapy. Instructions for getting that working are here: https://groups.io/g/sdrangel/message/1473

srcejon, 2020-09-25:
There are a few other issues:

If you start acquisition, stop it, then restart, SDRangel will crash.
Trying to change the sample rate while acquisition is running will cause SDRangel to crash.
When moving the "global gain" slider, it sometimes resets to 0 by itself.

These problems do not seem to occur when SoapySDR is used to control RTL SDR.

srcejon, 2020-10-03:
As SoapySDR support is a bit unreliable, I've written some native plugins to support USRP radios:

https://github.com/f4exb/sdrangel/pull/652

f4exb, 2020-10-06:
Thus superseded by #652 Implemented since 4.19.0 and 5.13.0

closed 2020-10-06 by f4exb