f4exb/sdrangel #618: ATV Demod Line and H Pulse sliders missing in 4.15.2

srcejon, 2020-08-26:
In SDRangel 4.15.2, it seems the ATV Demodulator Line and H Pulse sliders are missing:

[image]

They are there in the 4.15.0 release.

Without being able to adjust Line timing, I can't receive a stable image, as it seems Line always needs to be adjusted slightly from the default value. (I'm transmitting with SDRangel ATV modulator. E.g. 625 5FPS PAL625 needs adjusting from 320us to 315us)

srcejon, 2020-08-26:
I just noticed the 4.15.1 release note relating to the ATV Demod changes, states "Some controls and displays became obsolete". If the Line slider is intentionally obsolete, why might I not be receiving an image correctly? Here's a screenshot showing both Mod and Demod settings. 

[image]

Thanks.

f4exb, 2020-08-27:
Sorry it works and line length tweaking is completely useless this is why it was removed.

You have to check your setup or use different hardware. 

Tx is HackRF wti same sample rate:
[image]

Rx is Lime mini same sample rate:
[image]

srcejon, 2020-08-28:
Thanks f4exb, seeing all of your settings have helped getting it working. For anyone else that might read this, the settings that needed to be adjusted from defaults were:

Uniform level luminance 50 -> 100
Sync 150mV -> 100mV

closed 2020-08-28 by srcejon