f4exb/sdrangel #625: API devicesetChannelDelete channel 0 doesn't appear to work correctly if there are multiple channels

srcejon, 2020-09-05:
I would like to delete a channel using the API.

In the documentation for devicesetChannelDelete (http://127.0.0.1:8091/doc/swagger-ui/index.html#/DeviceSet/devicesetChannelDelete) it says: "delete channel (server only)". 

Is the "(server only)" bit a mistake? I presume we should be able to delete channels in the GUI? It appears to work if there is a single channel. 

There seems to be a bit of an issue if there are multiple channels however. If there are two channels, you can delete channel 1 then channel 0 successfully. However, if you have two channels, and try to delete channel 0 first, SDRangle will say there are no channels left when trying to delete the second channel (Even though one will still be displayed in the GUI).

E.g:

Start SDRangel:

curl -X POST "http://127.0.0.1:8091/sdrangel/deviceset/0/channel" -H  "accept: application/json" -H  "Content-Type: application/json" -d "{  \"channelType\": \"AMDemod\",  \"direction\": 0,  \"originatorDeviceSetIndex\": 0    }"
{
    "message": "Message to add a channel (MsgAddChannel) was submitted successfully"
}

curl -X POST "http://127.0.0.1:8091/sdrangel/deviceset/0/channel" -H  "accept: application/json" -H  "Content-Type: application/json" -d "{  \"channelType\": \"AMDemod\",  \"direction\": 0,  \"originatorDeviceSetIndex\": 0    }"
{
    "message": "Message to add a channel (MsgAddChannel) was submitted successfully"
}

curl -X DELETE "http://127.0.0.1:8091/sdrangel/deviceset/0/channel/0" -H  "accept: application/json"
{
    "message": "Message to delete a channel (MsgDeleteChannel) was submitted successfully"
}

curl -X DELETE "http://127.0.0.1:8091/sdrangel/deviceset/0/channel/0" -H  "accept: application/json"
{
    "message": "There is no channel at index 0. There are 0 channels"
}

f4exb, 2020-09-10:
There's a glitch in the deletion handling:
```
2020-09-10 18:59:25.927 (D) HttpConnectionHandler (0x55f1fd2716f0): received request from 127.0.0.1 (DELETE) /sdrangel/deviceset/0/channel/0                                                                                                
2020-09-10 18:59:25.928 (D) WebAPIAdapterGUI::devicesetChannelDelete: index: 0 nbChannels: 2                                                                                                                                                
2020-09-10 18:59:25.928 (D) MainWindow::handleMessages: message: MainWindow::MsgDeleteChannel                                                                                                                                               
2020-09-10 18:59:25.928 (D) DeviceUISet::deleteChannel: delete channel [sdrangel.channel.amdemod] at 0                                                                                                                                      
2020-09-10 18:59:25.928 (D) DSPDeviceSourceEngine::removeSink:  AMDemod                                                                                                                                                                     
2020-09-10 18:59:25.928 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink                                                                                                                                  
2020-09-10 18:59:25.928 (D) HttpConnectionHandler (0x55f1fd2716f0): disconnected                                                                                                                                                            
2020-09-10 18:59:25.928 (D) AudioDeviceManager::removeAudioSink: 0x55f1fd60d088                                                                                                                                                             
                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                            
2020-09-10 18:59:40.159 (D) HttpConnectionHandler (0x55f1fd2716f0): received request from 127.0.0.1 (DELETE) /sdrangel/deviceset/0/channel/0                                                                                                
2020-09-10 18:59:40.159 (D) WebAPIAdapterGUI::devicesetChannelDelete: index: 0 nbChannels: 0                                                                                                                                                
2020-09-10 18:59:40.159 (D) HttpConnectionHandler (0x55f1fd2716f0): disconnected 
```

f4exb, 2020-09-10:
In deviceuiset.cpp:
```c++
void DeviceUISet::deleteChannel(int channelIndex)
{
    if ((channelIndex >= 0) && (channelIndex < m_channelInstanceRegistrations.count()))
    {
        qDebug("DeviceUISet::deleteChannel: delete channel [%s] at %d",
                qPrintable(m_channelInstanceRegistrations[channelIndex].m_channelName),
                channelIndex);
        m_channelInstanceRegistrations[channelIndex].m_gui->destroy();
        m_channelInstanceRegistrations.removeAt(channelIndex);
        renameChannelInstances();
    }
}
```
and `m_gui->destroy()` calls the plugin gui destructor for example `~AMDemod()`:
```c++
AMDemodGUI::~AMDemodGUI()
{
    m_deviceUISet->removeRxChannelInstance(this);
    delete m_amDemod;
    delete ui;
}
```
The destructor removes the instance effectively (it uses `this`) but the next extra ` m_channelInstanceRegistrations.removeAt(channelIndex);` deletes blindly at the given index. Eventually this results in one extra plugin being deleted. The call to the destructor in `deviceuiset.cpp:` should be enough.

f4exb, 2020-09-13:
Implemented in v4.15.5 and v5.10.1

closed 2020-09-13 by f4exb