f4exb/sdrangel #630: Bladerf micro2.0 tx gain range mismatch

Abraham403, 2020-09-12:
Thank you so much for this great mutliplatform, multihardware, rich feature software, i truly think its one of its kind out there.

Recently i was playing with bladerf micro2.0 on using sdrAngel on windows and found it there is a. Ismatch between the gain slider range and the range supported by bladerf itself and because of this the output tx power is very low compared o what bladerf is capable off. I think i found whats missing in the code.
 it has to do with the integration with the libbladerf dll. it seems that are using old commands that you have deprecated as per this link https://www.nuand.com/libbladeRF-doc/v2.2.1/deprecated.html. in the source code you are using the commands bladerf_get_gain_range() and bladerf_set_gain() to set the global gain value, however the  GUI gain slider takes a range of -90dB to 0dB which is the range of the DSA stage not the global range, thats why when the slider is at 0dB its doesnt turn out to be a 0dB setting for dsa but for the total system gain which actually ranges from -23.75 to 66dB as per the output i got from the bladerf-cli (print gain command). so when the slider is at 0dB, the total gain is 0dB not the dsa stage gain which is wrong and that why the maximum tx with sdrangel is just -58dbm not +7dbm.

Thanks in advance for your help fixing the code
Wish you all the best

f4exb, 2020-09-14:
> it seems that are using old commands that you have deprecated

I don't understand... from the very documentation you quote `bladerf_get_gain_range()` is indeed the function to be used. In the function documentation it says "This may vary depending on the configured frequency, so it should be checked after setting the desired frequency" so this could be an issue since I think the range setting is done once and for all at plugin instance construction.

Abraham403, 2020-09-14:
Hi Edouard, 

Yes this has to do with the range offsetting, i think the below suggestion will fix it

--- a/plugins/samplesink/bladerf2output/bladerf2outputgui.cpp
+++ b/plugins/samplesink/bladerf2output/bladerf2outputgui.cpp
@@ -66,8 +66,8 @@ BladeRF2OutputGui::BladeRF2OutputGui(DeviceUISet *deviceUISet, QWidget* parent)
 
     m_sampleSink->getGlobalGainRange(min, max, step);
     qDebug("BladeRF2OutputGui::BladeRF2OutputGui: getGlobalGainRange: [%d,%d] step: %d", min, max, step);
-    ui->gain->setMinimum((min-max)/1000);
-    ui->gain->setMaximum(0);
+    ui->gain->setMinimum(min/1000);
+    ui->gain->setMaximum(max/1000);
     ui->gain->setPageStep(1);
     ui->gain->setSingleStep(1);
 
@@ -176,8 +176,8 @@ bool BladeRF2OutputGui::handleMessage(const Message& message)
         blockApplySettings(true);
         int min, max, step;
         m_sampleSink->getGlobalGainRange(min, max, step);
-        ui->gain->setMinimum((min-max)/1000);
-        ui->gain->setMaximum(0);
+        ui->gain->setMinimum(min/1000);
+        ui->gain->setMaximum(max/1000);
         ui->gain->setPageStep(1);
         ui->gain->setSingleStep(1);
         displaySettings();
@@ -188,8 +188,8 @@ bool BladeRF2OutputGui::handleMessage(const Message& message)
     else if (BladeRF2Output::MsgReportGainRange::match(message))
     {
         const BladeRF2Output::MsgReportGainRange& cfg = (BladeRF2Output::MsgReportGainRange&) message;
-        ui->gain->setMinimum((cfg.getMin()-cfg.getMax())/1000);
-        ui->gain->setMaximum(0);
+        ui->gain->setMinimum(cfg.getMin()/1000);
+        ui->gain->setMaximum(cfg.getMax()/1000);
         ui->gain->setSingleStep(1);
         ui->gain->setPageStep(1);

Abraham403, 2020-09-14:
Hi Edouard

Yes it Has to do with gain range offsetting, the below suggestions may help to fix it

--- a/plugins/samplesink/bladerf2output/bladerf2outputgui.cpp
+++ b/plugins/samplesink/bladerf2output/bladerf2outputgui.cpp
@@ -66,8 +66,8 @@ BladeRF2OutputGui::BladeRF2OutputGui(DeviceUISet *deviceUISet, QWidget* parent)
 
     m_sampleSink->getGlobalGainRange(min, max, step);
     qDebug("BladeRF2OutputGui::BladeRF2OutputGui: getGlobalGainRange: [%d,%d] step: %d", min, max, step);
-)ui->gain->setMinimum((min-max)/1000);
-)ui->gain->setMaximum(0);
+)ui->gain->setMinimum(min/1000);
+)ui->gain->setMaximum(max/1000);
     ui->gain->setPageStep(1);
     ui->gain->setSingleStep(1);
 
@@ -176,8 +176,8 @@ bool BladeRF2OutputGui::handleMessage(const Message& message)
         blockApplySettings(true);
         int min, max, step;
         m_sampleSink->getGlobalGainRange(min, max, step);
-)ui->gain->setMinimum((min-max)/1000);
-)ui->gain->setMaximum(0);
+)ui->gain->setMinimum(min/1000);
+)ui->gain->setMaximum(max/1000);
         ui->gain->setPageStep(1);
         ui->gain->setSingleStep(1);
         displaySettings();
@@ -188,8 +188,8 @@ bool BladeRF2OutputGui::handleMessage(const Message& message)
     else if (BladeRF2Output::MsgReportGainRange::match(message))
     {
         const BladeRF2Output::MsgReportGainRange& cfg = (BladeRF2Output::MsgReportGainRange&) message;
-)ui->gain->setMinimum((cfg.getMin()-cfg.getMax())/1000);
-)ui->gain->setMaximum(0);
+)ui->gain->setMinimum(cfg.getMin()/1000);
+)ui->gain->setMaximum(cfg.getMax()/1000);
         ui->gain->setSingleStep(1);
         ui->gain->setPageStep(1);

f4exb, 2020-09-14:
This is what is returned by the board:
```
2020-09-14 13:24:07.220 (D) BladeRF2OutputGui::BladeRF2OutputGui: getGlobalGainRange: [-23750,66000] step: 250
```
It appears to be in 1000th of dB. This is for display but the slider step should be based on step size.

Abraham403, 2020-09-14:
Exactly! When i console to the board it gives me the following gain range (-23.75 to 66dB). With the current SDRangel code, when i take the gain slider to maximum possible(0dB), tx power measured is -58dbm however when i manually set the gain to 66dB using the bord’s console, the power measured is +7dBm. So there is additional 66dB power missing with the current code.

f4exb, 2020-09-18:
Implemented in v4.16.1 and v5.11.1

closed 2020-09-18 by f4exb