f4exb/sdrangel #633: lost my glasses - upgrading happens how

marxenegls, 2020-09-14:
how does one efficiently upgrade sdrangle ?
do i need to rebuild the whole or just src/sdrangle   itself .. ??
but the build cache isn't there , right ???
so i point to the previous build dir ???? , no way !! voodoo  
this time round i'd like to add the pluto-sdr which i skipped on the previous build
....
...
i still haven't fund found my glasses , 300 euros worth

f4exb, 2020-09-14:
I wish you will retrieve your expensive glasses but I do not see the point of this ticket. Please be more specific else I will close it.

marxenegls, 2020-09-15:
upgrade. do i go through the whole build again to install the latest or just sdrangel src and point cmake to include dir where all the hw modules and libs are ?
Efficiently upgrade ,.. i must have gone over the docs a dozen times and haven't seen anything pertaining the procedure.
I assume it ll be a handy bit of added documentation for everyone .
i didn't have my glasses at hand earlier and few beers so excuse my ranting

f4exb, 2020-09-15:
Most people do it successfully with this procedure: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux
There is documentation it is not my responsibility if you cannot read and apply it.

closed 2020-09-15 by marxenegls