f4exb/sdrangel #634: New Feature suggestion: AFC & Beacon locking

Abraham403, 2020-09-16:
Hi Edouard,

I highly recommend adding these RX fearure to SDRangel,

AFC: Automatic frequency correction
This feature is used by amateurs who track FM weather and communication satelittes. What this does is it somehow tracks the FM signal as it shift slightly left or right due to doppler effect as the satellite approaches and departs away. It is very important to maintain the highest RX SNR possible.(available in SDRuno)

Beacon locking: this rx feature is a great have feature for satellite amateurs communicating on the geostationary satellite QO100. It locks the Rx frequency using the qo100 telemetry beacon frequency as a reference to compensate for drift / offset due to a poor consumer LNB with unstable tcxo, so you can listen to a certain frequency in a stable way even if the lnb tcxo is drifting. I think i know how this works exactly and can share with you the algorithm if you are willing to code it(available in sdrconsole)

Best Regards
Abraham

f4exb, 2020-09-16:
You can already do this using the http API. This is again a case of something that could be done in a distributed fashion but where people look for an integrated fashion (like SDRUno, SDR#, SDRConsole...). But this is what the Feature plugins are designed for so this request is valid. 

The [Frequency tracker](https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/freqtracker/readme.md) plugin does the tracking. It has FLL and PLL capabilities and for the PLL can lock to no modulation (thus same as FLL), BPSK, QPSK, 8-PSK. Its job is just to move its shift frequency to follow the signal to which it is locked. Because all plugins have so called "reverse API" they can send information in the same format as the incoming REST API whenever their state change to a network end point (address and port). Thus another piece of software can listen to this address and send back commands to the SDRangel instance using its incoming REST API. This is what this [Python script does](https://github.com/f4exb/sdrangel/blob/master/scriptsapi/freqtracking.py)

So in fact this functionality already exists but you have to assemble pieces together and this is what is not well understood or people are too lazy. The distributed approach means all pieces need not to be part of a big monolithic piece of software but could be spread over a network and this is eventually a more powerful and versatile architecture. But people especially ham radio operators do like big monoliths. So be it...

f4exb, 2020-11-01:
Implemented in v4.21.0 and v5.15.0

closed 2020-11-01 by f4exb