f4exb/sdrangel #635: Feature plugin: simple PTT

f4exb, 2020-09-16:
This is to re-implement [the PTT active script](https://github.com/f4exb/sdrangel/blob/master/scriptsapi/ptt_active.py) in a feature plugin with more or less the same functionality (may be less to keep it "simple"). Its first purpose is to exercise and tune the feature plugins framework with a very simple plugin.

Radiomix2000, 2020-09-16:
It would be really cool to add some hardware PTT functinality as well (using serial COM port for PTT is the most simplest way). So using PTT pedal will become again actual :)

f4exb, 2020-09-16:
But this is not "simple" anymore so I will keep this one under the KISS rule. Later on all sort of crap could be added to a "complex" PTT.

f4exb, 2020-09-28:
Implemented for v5 branch in v5.12.0

f4exb, 2020-09-29:
Done in v4.18.0

closed 2020-09-29 by f4exb