f4exb/sdrangel #637: DATV Decoder Delta F

m0dts, 2020-09-18:
Small bug regarding 'Delta F' for DATV Demdulator.

If i use the mouse to select the area of spectrum for the decoder the 'Delta F' is updated in the DATV demodulator window but the demodulator does not respond and i don't decode the signal.
If i then adjust the 'Delta F' tuning with the mouse on the DATV demodulator window the demodulator responds and starts to decode the signal.
Missing action when the spectrum is clicked to adjust decoder?

Thanks
Rob

f4exb, 2020-09-18:
You cannot "select an area" of the spectrum all you can do is point and click and the center frequency of the demodulator is move to that point.

This is only in the case there is only one demodulator. If you have more than one in the passband (channels window) then this functionality is disabled. This is because there is no indication of which channel to move. 

This is only a facility for single channel setups. You may always drag the center line of any channel to obtain the same result.

m0dts, 2020-09-18:
Ok thanks, sorry i did not mean select area but just click.

I understand the problem if multiple demodulators, not easy to implement.
In single channel mode this should work? 

Rob

f4exb, 2020-09-18:
Yes and I just verified this for DATV plugin.

m0dts, 2020-09-18:
Thanks, 
Here it does not work as expected...
The frequency is updated on the demodulator panel display when i click the spectrum but the decoder does not see the change.
If i then adjust the frequency by just 1Hz on the demodulator panel it then responds and locks onto the signal.
I'm using win x64 4.16.1.

f4exb, 2020-09-18:
Which signal is it? DVB-S or DVB-S2? and with which characteristics S/R, modulation, FEC?

m0dts, 2020-09-18:
It happens on all modes/sr/fec.
The constellation is just noise until i change the channel frequency manually by just a small amount, then i see the constellation appear.

m0dts, 2020-09-18:
I've made a short video here to explain:
https://www.dropbox.com/s/rz7t5k1cie6mxci/2020-09-18%2018-27-00.mp4?dl=0

f4exb, 2020-09-21:
I found that the point and click does not actually changes the plugin shift frequency. This is probably the issue. Of course tweaking the thumb wheels will fix this because then the face frequency gets correctly set.

f4exb, 2020-09-29:
Fixed in v4.17.0

closed 2020-09-29 by f4exb