f4exb/sdrangel #640: Add Channels window is malformed in 5.11.1

akhepcat, 2020-09-19:
Just built  5.11.1  Qt 5.12.8 x86_64  on (k)Ubuntu 20.04.1

Clicking on the '+'Channels  opens up a new window, with an "Available Channels"  selector.
The selector is malformed, at approximately 1/4 height of the text.

Clicking on the small, but visible portion opens the selector correctly, with non-garbled text.

f4exb, 2020-09-19:
I cannon reproduce the issue in Lubuntu (LxQt) 20.04 so this is installation specific and cannot be corrected.

akhepcat, 2020-09-19:
(sigh)     

So, I looked a little further, and found that changing one line solves this issue.

--- channeladddialog.ui~        2020-09-18 18:01:31.000000000 -0800
+++ channeladddialog.ui 2020-09-19 08:53:52.411290296 -0800
@@ -25,7 +25,7 @@
      <property name="maximumSize">
       <size>
        <width>16777215</width>
-       <height>70</height>
+       <height>170</height>
       </size>
      </property>
      <property name="font">

Hard-coding height-width properties instead of determining them at runtime based on font-size scaling is why this is happening.  Which means it's a deeper issue.

But at least I know *why*  it's occurring, and where the problem is.

f4exb, 2020-09-19:
This is because I use Qt Creator so the problem is rather with Qt Creator but there is no way I will hand craft everything:
[image]

akhepcat, 2020-09-19:
Ah - I wonder if it's a HighDPI issue?

https://doc.qt.io/qt-5/highdpi.html

My screen is:
screen #0:
  dimensions:    3840x2160 pixels (621x341 millimeters)
  resolution:    157x161 dots per inch

f4exb, 2020-09-21:
Yes there are issues with high DPI screens but possibly as you mention because of some settings in the .ui files. In fact in this case this comes from a Maximum Size / Height of 70 for this "GroupBox" layout that is not really neccessary. The Minimum size is used extensively though to ensure the widgets are big enough but the Maximum size constraint more rarely.

akhepcat, 2020-09-21:
well, if the "maximumSize"  property isn't used, then ...

    $ sed -i.bak '/<property name="maximumSize">/,/<\/property>/d'  $(find . -iname '*.ui' -exec grep -il maximumsize {} \;)

would remove all of them from the ui  files...

Guess i'll have to try that locally and see what breaks.

akhepcat, 2020-09-21:
Well, that was certainly an interesting test.  It's... Mostly better.   Though the volume controls in demods  should probably not lose that max-size property, otherwise they grow a bit too big, hahahah.

Probably not really worth  putting together an allow/deny workflow to remove unnecessary properties vs. essential max-size limits.    

Just a note in the wiki/readme  to use the  AUTO SCALE from above  when running sdrangel will probably suffice.

f4exb, 2020-09-22:
Good point for the Wiki

f4exb, 2020-10-08:
Done in the Wiki: https://github.com/f4exb/sdrangel/wiki/High-DPI-displays

closed 2020-10-08 by f4exb