f4exb/sdrangel #642: Highpass and bandpass filters have large sample rate dependent gain

srcejon, 2020-09-20:
The highpass and bandpass filters (in sdrbase/dsp/highpass.h and bandpass.h) appear to have large sample rate dependent gain in the pass band. This can be seen in the following frequency response plots for the high pass filter, based on the filter's taps, computed for:

    Highpass<Real> filter;
    filter.create(301, 48000, 5000);
    filter.create(301, 4*48000, 5000);

[image]

This appears to be due to the normalisation code using the value of the centre tap that has had 1 added to it on line 39 (that flips it from low pass to highpass).

It looks like this can be fixed, by simply not including this extra 1 in the normalisation:

Change:
Line 49: sum += m_taps[i];
To:
sum += m_taps[i] - 1;

This then gives 0dB gain in the passband, regardless of sample rate, and attenuates the stop band:

[image]

bandpass.h needs the same fix on line 70.

f4exb, 2020-09-21:
The Highpass is not used anywhere but the Bandpass in the following plugins. These plugins probably need to be checked because some hack was probably used to circumvent the issue that can now have an adverse effect:

- AM Demod
- NFM Demod
- UDP Sink
- NFM Modulator

f4exb, 2020-09-21:
In the NFM Demod there is a lowpass and bandpass filter created like this:
```cpp
        m_bandpass.create(301, m_audioSampleRate, 300.0, settings.m_afBandwidth);
        m_lowpass.create(301, m_audioSampleRate, settings.m_afBandwidth);
``` 
Then it is pretty obvious that it should be scaled by the number of taps:
```c++
                if (m_settings.m_highPass) {
                    sample = m_bandpass.filter(m_squelchDelayLine.readBack(m_squelchGate)) * m_settings.m_volume;
                } else {
                    sample = m_lowpass.filter(m_squelchDelayLine.readBack(m_squelchGate)) * m_settings.m_volume * 301.0f;
                }
```

f4exb, 2020-09-29:
Implemented in v4.17.0

closed 2020-09-29 by f4exb