f4exb/sdrangel #657: Version 5.14 nothing moving!!!

loewal, 2020-10-11:
Hi Edouard,

I had to revert to a nice working version 5.13.1
Neither waterfall nor spectrum moves. Compiled 5.14 on iMac an Ubuntu.
Same result.
????

f4exb, 2020-10-11:
Github issues are not an outlet. Can you describe the issue properly?

loewal, 2020-10-11:
Hi Edouard,

I don’t quite understand what you mean, but I’ll try:

Yesterday I compiled version 5.13.1 of SDRangel on two machines. They worked perfectly.
This afternoon (yes in Holland it was around 15:00 o’clock) I download version 5.14 and compiled.
Now on both machines the waterfall and spectrum don’t move whilst I can hear the sound.
Moving outside their windows, there is a short move, but then stops again.
With other words, no movement at all.

When reverting back to version 5.13.1, all went well again.

Thanks!

f4exb, 2020-10-11:
> Now on both machines the waterfall and spectrum don’t move whilst I can hear the sound.
Moving outside their windows, there is a short move, but then stops again.

Yes this describes the issue. It turns out that the master timer that has moved to a MainCore class is not started in the constructor. A fix has been promoted to v5 and tag v5.14.0 moved.

loewal, 2020-10-11:
Thank you!

Loe Walter
Insula 103
6416 BZ  Heerlen
☎️ +31-6-24476963

closed 2020-10-12 by f4exb