f4exb/sdrangel #659: receiver channels are not isolated

akhepcat, 2020-10-16:
I have a LimeSDR-USB,  which is mimo 2x2,  so I should be able to listen to two separate frequency ranges simultaneously.

I set  R0  up for some 2m decodes  on  Lime RX1

I then add  R1,  using  Lime RX2

Any changes to the tuning of  R0  immediately change  R1, which means I can't pay attention to both UHF and VHF channels.
 
Also, bookmarks don't differentiate, and overload both receivers with whatever was saved.

f4exb, 2020-10-16:
1. When you exit the program only the first device set (i.e. R0) is saved as default. This is for now normal behavior. It could possibly be made different but I don't think it is a good idea to have this as a default behavior. For now there are workarounds to further configure the instance by using the REST API. For this you can use this script: https://github.com/f4exb/sdrangel/tree/master/scriptsapi#configpy

2. In the Lime the two Rx channels share the same LO. You can only "differentiate" by using the NCO that can be set by channel. So again this is normal behavior. Same thing on the Tx side for both Tx channels. However the Rx and Tx subsystems do not share their LOs and thus Rx and Tx could be set wide apart.

akhepcat, 2020-10-16:
Ah, thanks for the clarification.    I hadn't yet tried config saving with multiple SDRs attached, but I did try loading bookmarks against the Lime  on R0   and my rtlsdr  on R1, and that seemed to work fine.

I'll have too look into the rest api more, i guess, if i want to be able to set up some interesting configs.

Thanks.

f4exb, 2020-11-15:
Question answered

closed 2020-11-15 by f4exb