f4exb/sdrangel #660: Waterfall on Windows can be missing

srcejon, 2020-10-16:
I've seen a few cases where the waterfall can be missing on Windows (once in a VM while release testing due to lack of Open GL support in graphics driver, and another while running SDRangel via Remote Desktop Protocol, where the waterfall appears when run locally, but not remotely).

Both cases appear to be fixed by installing Qt's ANGLE libraries (libEGL.dll et al), which maps OpenGL to DirectX, if full OpenGL support isn't available.

I've a patch for this, which I'll open a PR for shortly.

f4exb, 2020-10-16:
I thought Qt's ANGLE was coming in by default but I may have been mistaken

zildjiani, 2020-10-21:
73 F4EXB j'ai un probleme avec sdrangel sdrangel-4.20.0-win64.exe apres execution du programme 
la fenetre du spectrumanalyzer ne fonctionne pas:couleur blanc!
compiuter win10 familliale 64bits  processeur i3 6G ram

merci d'avance

flogger76, 2020-10-26:
Hello,
I have the same problem. I'v installed sdrangel-4.20.0-win64, and te effect is like on this picture:
[image]

I can't see the waterfall, spectrum, there is problem with refreshing frames inside the main window of program and sometime the program makes crashes.
My system:

Windows 7 64bit
uP Intel Core i5-2500 CPU 3,3GHz
graphic card Intel HD Graphic 2000, driver Intel v9.17.10.4229

Can you help me please what I have wrong ?

73!
Adam SQ1GPR

Davidi74, 2020-10-26:
Yes, waterfall is missing for me too in windows. I had reported(#513) in an earlier version also that the ATV decoder didn't refresh either, but the dev was unable to find the problem. Both issues occur on my two machines Which are windows 10 LTSC 2019

Davidi74, 2020-10-28:
Looks like this issue still persists on my Intel Iris based laptop with v4.21.0
Everything works fine on an my Nvidia 2080 based desktop.

I uninstalled the program on the intel iris based machine and found that the problem occurs with previous versions, however in  version 4.14.3 it works fine. So between 4.14.3 and 4.14.20 something appears to have changed.

f4exb, 2020-10-30:
Normally this is for Windows 10 not Windows 7 so anything can happen there. Anyway there is no much more that can be done.

closed 2020-10-30 by f4exb

Davidi74, 2020-10-30:
....Further to my previous post. Log file includes the following...

Failed to extract plugin meta data from 'C:/Program Files/SDRangel/plugins/libEGL.dll'

srcejon, 2020-10-30:
libEGL.dll should be in C:/Program Files/SDRangel not C:/Program Files/SDRangel/plugins

Did you move it yourself?

Davidi74, 2020-10-30:
Sorry! I must have while trying to see if I could get it working.
I have removed the file, and performed a fresh install to a different directory.

-----------
Attached log
----------

2020-10-31 09:25:10.372 (I) SDRangel 4.21.0 Qt 5.9.9 64b x86_64 Windows 10 (10.0) DSP Rx:24b Tx:16b PID 2700
2020-10-31 09:25:10.374 (D) MainWindow::MainWindow: load plugins...
2020-10-31 09:25:10.374 (D) PluginManager::loadPluginsPart folder C:/sdr/../lib/sdrangel/plugins is empty
2020-10-31 09:25:10.374 (D) PluginManager::loadPluginsPart folder C:/sdr/lib/plugins is empty
2020-10-31 09:25:10.374 (D) PluginManager::loadPluginsDir: fileName: chanalyzer.dll
2020-10-31 09:25:10.375 (I) PluginManager::loadPluginsDir: loaded plugin chanalyzer.dll
2020-10-31 09:25:10.375 (D) PluginManager::loadPluginsDir: fileName: demodadsb.dll
2020-10-31 09:25:10.377 (I) PluginManager::loadPluginsDir: loaded plugin demodadsb.dll
2020-10-31 09:25:10.377 (D) PluginManager::loadPluginsDir: fileName: demodam.dll
2020-10-31 09:25:10.377 (I) PluginManager::loadPluginsDir: loaded plugin demodam.dll
2020-10-31 09:25:10.382 (D) PluginManager::loadPluginsDir: fileName: demodatv.dll
2020-10-31 09:25:10.382 (I) PluginManager::loadPluginsDir: loaded plugin demodatv.dll
2020-10-31 09:25:10.382 (D) PluginManager::loadPluginsDir: fileName: demodbfm.dll
2020-10-31 09:25:10.383 (I) PluginManager::loadPluginsDir: loaded plugin demodbfm.dll
2020-10-31 09:25:10.383 (D) PluginManager::loadPluginsDir: fileName: demoddatv.dll
2020-10-31 09:25:10.385 (I) PluginManager::loadPluginsDir: loaded plugin demoddatv.dll
2020-10-31 09:25:10.385 (D) PluginManager::loadPluginsDir: fileName: demoddsd.dll
2020-10-31 09:25:10.386 (I) PluginManager::loadPluginsDir: loaded plugin demoddsd.dll
2020-10-31 09:25:10.386 (D) PluginManager::loadPluginsDir: fileName: demodfreedv.dll
2020-10-31 09:25:10.387 (I) PluginManager::loadPluginsDir: loaded plugin demodfreedv.dll
2020-10-31 09:25:10.387 (D) PluginManager::loadPluginsDir: fileName: demodlora.dll
2020-10-31 09:25:10.388 (I) PluginManager::loadPluginsDir: loaded plugin demodlora.dll
2020-10-31 09:25:10.388 (D) PluginManager::loadPluginsDir: fileName: demodnfm.dll
2020-10-31 09:25:10.388 (I) PluginManager::loadPluginsDir: loaded plugin demodnfm.dll
2020-10-31 09:25:10.388 (D) PluginManager::loadPluginsDir: fileName: demodssb.dll
2020-10-31 09:25:10.389 (I) PluginManager::loadPluginsDir: loaded plugin demodssb.dll
2020-10-31 09:25:10.389 (D) PluginManager::loadPluginsDir: fileName: demodwfm.dll
2020-10-31 09:25:10.389 (I) PluginManager::loadPluginsDir: loaded plugin demodwfm.dll
2020-10-31 09:25:10.389 (D) PluginManager::loadPluginsDir: fileName: featureafc.dll
2020-10-31 09:25:10.390 (I) PluginManager::loadPluginsDir: loaded plugin featureafc.dll
2020-10-31 09:25:10.390 (D) PluginManager::loadPluginsDir: fileName: featuregs232controller.dll
2020-10-31 09:25:10.390 (I) PluginManager::loadPluginsDir: loaded plugin featuregs232controller.dll
2020-10-31 09:25:10.390 (D) PluginManager::loadPluginsDir: fileName: featurerigctlserver.dll
2020-10-31 09:25:10.391 (I) PluginManager::loadPluginsDir: loaded plugin featurerigctlserver.dll
2020-10-31 09:25:10.391 (D) PluginManager::loadPluginsDir: fileName: featuresimpleptt.dll
2020-10-31 09:25:10.391 (I) PluginManager::loadPluginsDir: loaded plugin featuresimpleptt.dll
2020-10-31 09:25:10.391 (D) PluginManager::loadPluginsDir: fileName: filesink.dll
2020-10-31 09:25:10.392 (I) PluginManager::loadPluginsDir: loaded plugin filesink.dll
2020-10-31 09:25:10.392 (D) PluginManager::loadPluginsDir: fileName: filesource.dll
2020-10-31 09:25:10.392 (I) PluginManager::loadPluginsDir: loaded plugin filesource.dll
2020-10-31 09:25:10.392 (D) PluginManager::loadPluginsDir: fileName: freqtracker.dll
2020-10-31 09:25:10.393 (I) PluginManager::loadPluginsDir: loaded plugin freqtracker.dll
2020-10-31 09:25:10.393 (D) PluginManager::loadPluginsDir: fileName: inputairspy.dll
2020-10-31 09:25:10.394 (I) PluginManager::loadPluginsDir: loaded plugin inputairspy.dll
2020-10-31 09:25:10.394 (D) PluginManager::loadPluginsDir: fileName: inputairspyhf.dll
2020-10-31 09:25:10.394 (I) PluginManager::loadPluginsDir: loaded plugin inputairspyhf.dll
2020-10-31 09:25:10.394 (D) PluginManager::loadPluginsDir: fileName: inputbladerf1.dll
2020-10-31 09:25:10.395 (I) PluginManager::loadPluginsDir: loaded plugin inputbladerf1.dll
2020-10-31 09:25:10.395 (D) PluginManager::loadPluginsDir: fileName: inputbladerf2.dll
2020-10-31 09:25:10.396 (I) PluginManager::loadPluginsDir: loaded plugin inputbladerf2.dll
2020-10-31 09:25:10.396 (D) PluginManager::loadPluginsDir: fileName: inputfileinput.dll
2020-10-31 09:25:10.396 (I) PluginManager::loadPluginsDir: loaded plugin inputfileinput.dll
2020-10-31 09:25:10.396 (D) PluginManager::loadPluginsDir: fileName: inputhackrf.dll
2020-10-31 09:25:10.397 (I) PluginManager::loadPluginsDir: loaded plugin inputhackrf.dll
2020-10-31 09:25:10.397 (D) PluginManager::loadPluginsDir: fileName: inputkiwisdr.dll
2020-10-31 09:25:10.397 (I) PluginManager::loadPluginsDir: loaded plugin inputkiwisdr.dll
2020-10-31 09:25:10.397 (D) PluginManager::loadPluginsDir: fileName: inputlimesdr.dll
2020-10-31 09:25:10.398 (I) PluginManager::loadPluginsDir: loaded plugin inputlimesdr.dll
2020-10-31 09:25:10.398 (D) PluginManager::loadPluginsDir: fileName: inputlocal.dll
2020-10-31 09:25:10.398 (I) PluginManager::loadPluginsDir: loaded plugin inputlocal.dll
[... cut ...]
2020-10-31 09:25:56.157 (D) DSPDeviceSourceEngine::gotoIdle
2020-10-31 09:25:56.157 (D) DSPDeviceSourceEngine::stop
2020-10-31 09:25:56.157 (D) DSPDeviceSourceEngine::gotoIdle
2020-10-31 09:25:56.163 (D) HttpListener: closed
2020-10-31 09:25:56.163 (D) HttpConnectionHandlerPool (0x1660b15d060): destroyed
2020-10-31 09:25:56.163 (D) HttpListener: destroyed
2020-10-31 09:25:56.163 (I) WebAPIServer::stop: stopped web API server at http://127.0.0.1:8091
2020-10-31 09:25:56.163 (D) MainWindow::~MainWindow: end
[image]

flogger76, 2020-10-31:
I'v installed Windows 10 special for SDRAngel and with version 4.20.0-win64 behaviored the same like in Win7. New version 4.21.0 works better (Windows 10 the same hardware like previously). The software now no makes crashes. There is no problem with displaying frames, I can see the spectrum, but no waterfall. There is black square. 
For me this is no problem because I need SDRAngel only to transmit with hackRF (QO-100), but it will be nice when the program will be full well :) .

Davidi74, 2020-11-01:
Try downloading version 4.14.3.

This seems to be the last version (or close it) that the waterfall works. If it also works for you, it might help the developer see the  issue.

flogger76, 2020-11-03:
On the github I found the oldest version binary 4.18.0. I tried with newest 4.21.1 and is the same like previous - no waterfall.

Davidi74, 2020-11-03:
Lucky I keep things for just such an occasion.

Copy and paste the link, its an EXE file in a ZIP file.

https://www.dropbox.com/s/cmd6u2fxttu4b00/sdrangel-4.14.3-win64.zip?dl=0

The dev did mention that us, the users should try to be "level 1 support" so here is my effort to do so.

Im fairly sure its the intel graphics and something changed after this version. I don't particularly like the intel GPU but there is limited choice in smaller laptops which is what I mainly use.

Anyway please check and let me know if this fixes the problem and *maybe* help the dev.

flogger76, 2020-11-04:
With 4.14.3 is bad. It is similarly like with 4.20.0.

`Anyway please check and let me know if this fixes the problem and maybe help the dev find the issue :)`

Yes that's right. When you have any ideas, please give me too. :) .

srcejon, 2021-01-29:
Try setting the environment variable QT_OPENGL to software before running SDRangel.

srcejon, 2021-01-29:
When running with QT_OPENGL=angle, I get:

 GLSpectrum::initializeGL: context:  major:  2  minor:  0  ES:  yes

With QT_OPENGL=software, I get:

GLSpectrum::initializeGL: context:  major:  3  minor:  0  ES:  no

With angle, the waterfall and text labels are missing, whereas they appear with the software renderer, so I presume they are dependent upon an OpenGL 3 feature that isn't supported by the angle driver.

It looks like angle has been removed from Qt6, but software rendering will remain.

If we wanted an option in SDRangel to select the software renderer, it seems the QApplication attribute Qt::AA_UseSoftwareOpenGL can be set - but unfortunately, it doesn't look like you can get the major version of the default driver first, so SDRangel would need to be restarted after changing this setting. But at least we could popup a warning with some info of what to do, to fix it.

Davidi74, 2021-01-30:
Yes, that fixed it!

Thank you very much for this!

tomcarlson, 2023-10-30:
setting the environment variable QT_OPENGL to software also worked for me.  I know have waterfall on an older NUC running WIn 11 Pro.  Had no waterfall before.