f4exb/sdrangel #661: Test source does not appear to work

srcejon, 2020-10-21:
Hi,

It looks like the Test Source does not appear to work. If I press start acquisition, there's no spectrum/waterfall (or anything in channel analyser scope) regardless of test source settings. Have tried both on Windows and Linux with SDRangel 4.20.

chris4git, 2020-10-25:
Hi,

same with SDRangel 5.14.0/Ubuntu 20.04

f4exb, 2020-11-01:
I don't know for how long it has been like this but this seems to be due to the new worker handling:
```
2020-11-01 10:06:34.934 (D) TestSourceWorker::startWork
2020-11-01 10:06:34.934 (W) QObject::startTimer: Timers cannot be started from another thread
```
The solution is to move timer start and stop in constructor and destructor respectively. Stream generation is conditional to `m_running` anyway.

f4exb, 2020-11-03:
Implemented in v4.21.1 and 5.15.1

closed 2020-11-03 by f4exb