f4exb/sdrangel #662: sound interruptions in ssb demod

pe1iee, 2020-10-22:
I use SDRangel also to work on QO100 in ssb with the frequency tracker active.
Upgraded to 4.20 coming from 4.14 and I noticed that audio now stutters when tuning in ssb.
Also when the frequency tracker makes corrections. 
In 4.14 this was not the case and it was 'smooth' fine-tuning. 
Is this by design or is there a setting to correct this?
Please advise.
Peter, PE1IEE

f4exb, 2020-10-22:
> Also when the frequency tracker makes corrections.
I am sorry you cannot expect SSB audio to be stable if the frequency changes. Do you know how SSB works? 

Secondly the correction can do nothing if the microwave LO is wiggling which is happening on cheap LNBs. It can only correct a slow drift. I think you have to get yourself a better LNB. You also have to check the controls of the frequency tracker so that it does not attempt to correct all the time. See the note [here](https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/freqtracker/readme.md#2-instantateous-tracker-error)

pe1iee, 2020-10-22:
Thank you for your quick response! 
I am fully aware how SSB works in SDRs and just observed that now the audio is briefly muted while tuning.
This was not the case in the 4.14 version and that's why I was asking if this was changed since then.

The tuning I do with a trackerball that has a big tuning ring to shift up-and down in the active window.
If I place the cursor  in the spectrum dispay where the SSB demod is I can tune very smooth the SSB freq by turning the ring.
Works very nice as if tuning on a radio knob. 

You are completely right about the cheap LNB's: they wiggle too much and are useless for SSB. 
That's why I am using a proper PLL controlled one :)  
The tracker only compensates for slow temperature drift and does a very nice job!

Hope you can help me
Peter, PE1IEE

f4exb, 2020-10-24:
There are no issues noted with the SSB channel. It is normal that in addition to the effect due to frequency variation the audio is chopped when the SSB channel shift frequency is adjusted since there is a mutex mechanism to avoid a conflict with the DSP processing. It only becomes annoying when this happens all the time. Therefore you have to adjust parameters so that the drift readjustment is not triggered extensively.

pe1iee, 2020-10-25:
Thanks again for the explanation.  I understand that it is by design that the audio is now muted while tuning.
No problem; I will stick with the 4.14 release for working in SSB as there is no muting and is much more comfortable listening.
Keep up the good work on SDRangel as it is by far the best!
73 PE1IEE

closed 2020-10-29 by pe1iee