f4exb/sdrangel #664: CTCSS max at 203.5

mcyclone, 2020-10-24:
There are 55 standard CTCSS tones

67 - 254.1 however in sdrangle only had max 203.5 only, it good if range can add or using manual input is better too

f4exb, 2020-10-24:
There are multiple standards covering 51 tones: https://en.wikipedia.org/wiki/Continuous_Tone-Coded_Squelch_System
From which only 32 are implemented in SDRangel. So indeed it is missing 19 of them.
I don't think it is a good idea to the tone a variable.

f4exb, 2020-10-24:
Note: definition is here https://github.com/f4exb/sdrangel/blob/master/sdrbase/dsp/ctcssdetector.cpp#L21

texasyojimbo, 2020-10-26:
Hi Edouard,

Perhaps CTCSS tones could be defined in a text file or something instead of hard-coded. If you did it that way, users like mcyclone could add the specific tones they need. Or, if user only needs one CTCSS tone, for example monitoring a local repeater, user could get rid of the un-needed tones. It would allow people to use different standards just by replacing a text file...

f4exb, 2020-11-03:
Implemented in v4.21.1 and 5.15.1

closed 2020-11-03 by f4exb