f4exb/sdrangel #665: Frequency Tracker: spectrum display

f4exb, 2020-10-24:
It would be good to have a spectrum display in the Frequency Tracker UI showing the channel spectrum to see a little bit better what's going on.

srcejon, 2020-10-27:
I think it would be a nice feature to have if a user could add a spectrum or scope to any channel (both RX and TX).

f4exb, 2020-10-27:
I don't think this is useful for every channel type. Here it does have some added value because the lock condition is not very reliable. This could be improved as well but having a visual evidence on where the lock point is is more expressive. Like if you move the frequency slightly and that point doesn't move means that you do not have a lock effectively.

And then what data shall we monitor in this spectrum? If this is the portion of spectrum seen by the channel then superimposing a channel analyzer on top of it does exactly the same job.

Adding a spectrum display (or worse a scope display) has an impact on performance so I'd rather keep it where it is actually necessary to show something else than just a zoom on a portion of the global spectrum.

f4exb, 2020-11-01:
Implemented in v4.21.0 and v5.15.0

closed 2020-11-01 by f4exb