f4exb/sdrangel #668: Crash if adding Lime RX after TX

srcejon, 2020-10-26:
To reproduce :

Run SDRAngel 4.20.0
Set R0 to File Input
Add sink device set
Set T1 to LimeSDR[0:0]
Set R0 to LimeSDR[0:0]
Crashes on both Windows and Linux

f4exb, 2020-11-01:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f3e018a6107 in QLayout::addChildWidget(QWidget*) () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
[Current thread is 1 (Thread 0x7f3dfc832000 (LWP 3584230))]
(gdb) bt
#0  0x00007f3e018a6107 in QLayout::addChildWidget(QWidget*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007f3e018ac260 in QStackedLayout::insertWidget(int, QWidget*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007f3e01a5cd8a in QTabWidget::insertTab(int, QWidget*, QIcon const&, QString const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007f3e01a5ce5b in QTabWidget::insertTab(int, QWidget*, QString const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007f3e026f8421 in MainWindow::setDeviceGUI(int, QWidget*, QString const&, int) (this=0x7ffd3e85a5f0, deviceTabIndex=0, gui=0x7f3df4010100, deviceDisplayName=..., deviceType=0) at /opt/build/sdrangel/sdrgui/mainwindow.cpp:721
#5  0x00007f3e026ff2fe in MainWindow::sampleSourceChanged(int, int) (this=0x7ffd3e85a5f0, tabIndex=0, newDeviceIndex=4) at /opt/build/sdrangel/sdrgui/mainwindow.cpp:1888
#6  0x00007f3e026fea73 in MainWindow::samplingDeviceChanged(int, int, int) (this=0x7ffd3e85a5f0, deviceType=0, tabIndex=0, newDeviceIndex=4) at /opt/build/sdrangel/sdrgui/mainwindow.cpp:1786
```

f4exb, 2020-11-01:
I don't know how to fix this

f4exb, 2020-11-02:
Maybe that was too obvious :smile: because this is incorrect: https://github.com/f4exb/sdrangel/blob/master/sdrgui/mainwindow.cpp#L1716 as it always destroys the GUI in last position which will work only if you change the last device.

closed 2020-11-02 by f4exb

f4exb, 2020-11-03:
Implemented in v4.21.1 and 5.15.1

commit 5903e170 (2020-11-02): Main Window: fixed sampling device changed. Fixes #668
commit 25e0b5b2 (2020-11-02): Main Window: fixed sampling device changed. Fixes #668