f4exb/sdrangel #669: Scope MagSq trigger does not seem to work

srcejon, 2020-10-26:
Hi,

It appears that the MagSq trigger in the scope does not work. I can use Real and Mag trigger successfully, so I believe I'm using it correctly.

To reproduce:
- Open the scope with a signal coming in
- Set trace input to MagSq
- Set trigger source to MagSq
- Disable free run 
- Adjust trigger level to a point at which the trigger should occur

While the trigger level slider is moved, the scope will trigger and the display will be updated. However, after setting the level and releasing the slider, the scope appears to no longer trigger.

Testing on SDRangel 4.20 in both Windows and Linux.

f4exb, 2020-11-02:
This condition: https://github.com/f4exb/sdrangel/blob/master/sdrgui/dsp/scopevis.h#L1062 should be valid for `Projector::ProjectionMagSq` also.

closed 2020-11-02 by f4exb

f4exb, 2020-11-03:
Implemented in v4.21.1 and 5.15.1

commit 593d9551 (2020-11-02): Scope MagSq trigger fix. Fixes #669
commit 42d56a93 (2020-11-02): Scope MagSq trigger fix. Fixes #669