f4exb/sdrangel #672: UpChannelizer in demodulators

srcejon, 2020-10-28:
In the ADS-B demodulator (and perhaps future demodulators), it can be beneficial if the channel sample rate is higher than the baseband sample rate (particularly for SDRs such as the RTL-SDR, that are limited to ~2MSa/a, because the datarate is 1Mbit/s). 

Currently the Interpolator class is used for the upsampling, however, this probably has quite a high CPU overhead given it supports arbitrary rates and the high sample rate used. It would probably be better to use to the half-band interpolators in the UpChannelizer class. However, these are currently written only with a ChannelSampleSource interface (pull) without ChannelSampleSink (feed), which is needed for a demodulator. It therefore would be useful if the UpChannelizer supported both pull and feed, or there was a variant that did (if I haven't missed it).

f4exb, 2020-10-28:
Upsampling is useless in reception if used systematically. You will not gain any bandwidth because you simply cannot invent data that does not exist. Upsampling works correctly only if associated to filtering according to the original sample rate and therefore bandwidth. If for some reason interpolation has to take place because a specific sample rate is needed by a channel this should be done in the channel.  I suppose this is what the ADS-B demodulator does and anyway would work best if the input sample rate is higher than the requested sample rate. I think the DSD demodulator works the same way because it has to get enough samples to handle symbol processing (48 kS/s thus 10 samples / symbol at 4.8 kS/s data rate).

srcejon, 2020-10-28:
It's not about gaining bandwidth, but as you say, resampling to get a suitable sample rate for the demodulator.

Ideally the input sample rate would be higher, but the problem is that with ADS-B the data rate is 1Mbit/s, which means we want a sample rate several times this (E.g. 4 or 6MSa/s), but devices such as RTL-SDR, have a limit of ~2MSa/s. So to support these devices, we need interpolation (or a demodulator that works differently to account for sampling right at the Nyquist limit - but I think interpolation is easier).

At the moment, this is indeed done in the channel with the Interpolator class, but I suspect it would be more efficient to use half-band filters (and efficiency is important, given the high sample rate). I can write a new interpolator just for this if needed, but as you've already implemented what looks like a good version the UpChannelizer class, and it would potentially be useful in other digital demodulators that require high sample rates, I thought it would be good to try to reuse them.

f4exb, 2020-10-28:
I think I see better what you would like to have. In fact instead of having up and down channelizer let's just call it channelizer and it would adapt to whatever its input and output rates are. Requires a bit of work but feasible. I think using half band decimators exclusively is possible if the ratio of rates is a power of two. In this case it skips the rational decimator that becomes useless. I am sure this works for decimation in the Down Channelizer but not so sure for the other way round although I would have a closer look to the LocalSink and LocalSource channels.

srcejon, 2020-10-28:
Yep - a single channelizer that can go either way, only using half band filters if a power of two sounds ideal. 

There's no urgency as the code works as is, but I thought it would be a useful enhancement to have on the list. Thanks.

f4exb, 2020-10-28:
In the particular case of ADS-B if I'm right this is PPM (Pulse Position Modulation). For this particular modulation there could be (I am not saying there is...) tricks that could avoid brute force up sampling. I am wondering how the RTL-SDR based software like [dump1090](https://github.com/antirez/dump1090) does that may be worth having a look.

f4exb, 2020-10-29:
I had a second thought about this and while I retain the idea that merging the down and up channelizer into a single full featured channelizer would help code readability and possibly efficiency this is not the issue with ADS-B. I don't think that upsampling from 2 MS/s to 4MS/s to reach 2 samples per chip would bring a significant gain in detection sensitivity because as I said earlier at 2 MS/s the information is simply not there. It would be worth trying 2 MS/s and thus 1 sample per chip with RTL-SDR at 2 MS/s that is widely used for ADS-B monitoring.

The performance issue is different and I have logged an improvement that seems to work well (testing phase) here: #675

f4exb, 2020-11-17:
We still need 2 flavors of channelizers because each is specialized on a type of channel:

DownChannelizer:
```c++
class SDRBASE_API DownChannelizer : public ChannelSampleSink {
public:
	DownChannelizer(ChannelSampleSink* sampleSink);

UpChannelizer:
```c++
class SDRBASE_API UpChannelizer : public ChannelSampleSource {
public:
    UpChannelizer(ChannelSampleSource* sampleSource);
```

Therefore this can only be DownChannelizer -> SinkChannelizer and UpChannelizer -> SourceChannelizer each implementing decimation and interpolation.

f4exb, 2020-11-17:
Decimating on transmission does not make any sense so we could forget up/down sampling for the SourceChannelizer. On  reception it may make some sense for plugins with audio or DSD that require 48 kS/s.

f4exb, 2020-12-13:
I am not going to cary on with this since there is very little use of making upsampling available for Rx generally and not at all for downsampling for Tx.

closed 2020-12-13 by f4exb