f4exb/sdrangel #674: NFM demod issue

kasper93, 2020-10-29:
Hi,

I noticed that filtering is quite weak. I constantly hear CTCSS tone.

Recorded audio (CTCSS 156.7Hz, different recordings, but it doesn't mattet), few seconds of signal with some silence: 
High-pass filter enabled:
[image]

High-pass filter disabled:
[image]

While filtering is working the tone is still very audible, it makes little difference to actual output audio. Also I have set AFBW to 3k. Could we improve this situation? 

For example SDR# (without any filtering, just demod) produces perfectly clear audio without any noticeable sigh of this tone, look at this 300Hz cut-off:
[image]

And small GUI issue, CTCSS tone frequency is not displayed. It always stays at `--`. Squelch works fine if I select different freq, so just the current value doesn't seem to be passed to GUI.

f4exb, 2020-10-31:
Very obviously SDR# uses a FFT boxcar filter. Indeed there is a GUI feedback issue.

kasper93, 2020-10-31:
I fixed all things that bothered me. Will send PR later and we can discuss it further.

f4exb, 2020-10-31:
Fix on-going on my side along with extension of CTCSS codes (Issue #664)

kasper93, 2020-10-31:
Since I had some changes already done I pushed them. Feel free to do the changes your way :) Mostly minor adjustments in fact, but as I was debugging I refactored some parts also.

kasper93, 2020-11-01:
I guees, it is fixed now.

closed 2020-11-01 by kasper93

f4exb, 2020-11-03:
Implemented in v4.21.1 and 5.15.1