f4exb/sdrangel #677: the map is not displayed

choupette-change, 2020-10-29:
hello everyone, i just installed the latest version: 4.21.0. I am using the ADSB channel but my card remains desperately empty and white, where is the problem?
[image]

srcejon, 2020-10-29:
If this is on windows, it may be because the opensll libraries were not included in the first release. They should in the next build. You should also be able to copy them in yourself.

srcejon, 2020-10-29:
The libraries are here: https://github.com/f4exb/sdrangel-windows-libraries/tree/master/openssl

choupette-change, 2020-10-29:
Thanks for the answer but I'm on Ubuntu 20.04.01 LTS ...

srcejon, 2020-10-29:
Can you attach the debug log, please?

choupette-change, 2020-10-29:
ok, i'm new, where can i find it? I only upgraded the part https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux#build-sdrangel. Do I have to reinstall all ? https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

srcejon, 2020-10-29:
Under Preferences menu > Logging. Set log file to Debug and Check log to file. 

You might want to check you have the packages installed listed in the second link, particularly these, which are needed for the map:

libqt5quick5 \
qml-module-qtlocation  qml-module-qtlocation qml-module-qtpositioning qml-module-qtquick-window2 \
qml-module-qtquick-dialogs qml-module-qtquick-controls qml-module-qtquick-layouts \
libqt5serialport5-dev qtdeclarative5-dev qtpositioning5-dev qtlocation5-dev

choupette-change, 2020-10-29:
it's ok, it was missing:
qml-module-qtlocation
qml-module-qtpositioning
qml-module-qtquick-window2
qml-module-qtquick-dialogs
qml-module-qtquick-controls
qml-module-qtquick-layouts
qtlocation5-dev
I installed:
--- sudo apt-get update && sudo apt-get -y upgrade
--- sudo apt-get install libqt5quick5 qml-module-qtlocation qml-module-qtpositioning qml-module-qtquick-window2 qml-module-qtquick-dialogs qml-module-qtquick-controls qml-module-qtquick-layouts libqt5serialport5-dev qtdeclarative5-dev qtpositioning5-dev qtlocation5-dev
I reinstalled: https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux#build-sdrangel
Besides, can you tell me the optimum settings because I find it less sensitive than ADSBOX for example. Thank you
[image]

f4exb, 2020-10-30:
In the next release the correlation threshold setting is going to change a little bit (see #675) so what you may learn now may change a little bit later.

choupette-change, 2020-10-30:
Is it possible to output the information in SBS-1 format?

srcejon, 2020-10-30:
Not currently, just Beast format. What do you need SBS-1 for?

choupette-change, 2020-10-30:
I use it with the script on Xastir: Xastir / scripts / ads-b.pl

choupette-change, 2020-11-02:
How to change the units into SI units: Altitude ft in m, speed kn in km / h, climb etc ...? In which files to make the modifications? thank you for the info

f4exb, 2020-11-02:
> In which files to make the modifications?

There is no such think I guess. This has to be an option in the GUI. Although in aviation knots and feet are used it could be interesting to translate values to the metric system that may be more meaningful for people (like me) who have been dealing with meters since the childhood.

srcejon, 2020-11-02:
It doesn't currently, but will do in the next release, hopefully ready later this week.

srcejon, 2020-11-03:
The map isn't currently displayed on the Windows 4.21.1 release. 

There is an error in the log file: "QtQuick.Window" version 2.4 is not installed

This is because it's built using Qt 5.9.9 and according to https://doc.qt.io/qt-5/qtquickcontrols-index.html#versions Qt 5.9.x has version 2.2 of the QtQuick controls. Version 2.4 is only available with Qt >5.11. (I was using 5.13.2 for my development)

Is there any plan to upgrade Qt on Windows, or should I get it working with 5.9?

f4exb, 2020-11-03:
Windows binaries are produced using Appveyor images. See: https://www.appveyor.com/docs/windows-images-software/ It looks like Qt 5.13.2 should be available for VS 2017 image (that is used) but may require some definition in the `.appveyor.yml` file that I am not aware of.

f4exb, 2020-11-03:
Some contribution effort is being made to upgrade to MSVC 2019 and this image uses Qt 5.15.1. It looks like it builds. I am waiting for a PR.

f4exb, 2020-11-03:
This https://github.com/f4exb/sdrangel/blob/master/cmake/Modules/QtLocator.cmake#L10 has to be changed so that it works with 5.1x ranges:
```
FILE(GLOB QT_VERSIONS "${QT_BIN}/5.1*")
```
Then:
```
-- Qt found: C:/Qt/5.13.2/msvc2017_64
-- Compiling for 24 bit Rx DSP chain
```

choupette-change, 2020-11-04:
Following the installation of v4.21.1, the value 2MS/s 
[image]
does not appear ??
ADS-B: added 2MS / s sample rate

f4exb, 2020-11-04:
https://github.com/f4exb/sdrangel/commit/ecbf85e1a08e41477d310f06246d1bb9785359db

f4exb, 2020-11-08:
All mentioned extra pieces of functionality were implemented in this PR: #688

closed 2020-11-08 by f4exb