f4exb/sdrangel #681: SDRangel with 2 PlutoSDR

Radiomix2000, 2020-11-03:
Please help! Cannot start 2 plutos in SDRangel simultaneously. One Pluto starts, but the second doesn't.
No spectrum/waterfall, showing strangeactual gain values, no rssi, and no temp. The video below shows better than 1000 words what I mean.
What I'm doing wrong?
The version 4.11.9 works like expectedm but the last version from release page has these strange issues.
Both plutos are v0.32 firmware, but with v0.31 is the same situation.
https://www.youtube.com/watch?v=Lh0w0G70_sA

Radiomix2000, 2020-11-12:
Nobody has the same issue?

ChiefGokhlayeh, 2021-05-30:
Hi @Radiomix2000,
I too encounter this issue, even in the current latest release v6.13.0.

I did some digging and found that the segmentation fault is caused by an use-after-move bug in the Pluto device scan plugin.

https://github.com/f4exb/sdrangel/blob/1e5c4343d197e8d90642d2ee92d91eda85b76b7e/devices/plutosdr/deviceplutosdrscan.cpp#L67-L77
Here the scan results from libiio are pushed into a `std::vector` called `m_scans`. Next the device URI and serial number are extracted and stored in two separate `std::maps` called `m_uriMap` and `m_serialMap`. Note that `m_scans` allocates memory for each its `DeviceScan` elements, so the original design tried to save memory by pointing `m_uriMap` and `m_serialMap` to said memory via simple pointers. This is an issue, because each time the vector `m_scans` resizes all elements are moved out of the old buffer and into a new one. Thus all previously created pointers to its elements now point into deallocated memory. This causes a segmentation fault in [`DevicePlutoSDRScan::getURIFromSerial()`](https://github.com/f4exb/sdrangel/blob/1e5c4343d197e8d90642d2ee92d91eda85b76b7e/devices/plutosdr/deviceplutosdrscan.cpp#L104) when two devices are added.

@f4exb I implemented a small [fix](https://github.com/ChiefGokhlayeh/sdrangel/commit/c41fc66f3f1c5f183c8239e14653d0f6374f2c1a) for this bug and would like to help out! Do you want me to raise a PR? My fix consists of storing the `DeviceScan`s as `shared_ptr<DeviceScan>`, as to 1) don't rely on memory allocated by vector and 2) keep track of when it's safe to delete them.

PR #913 (2021-05-30): Use shared_ptr to keep track of DeviceScan across multiple collections
Fixes #681.

Relying on memory allocated by `std::vector` via direct pointers is dangerous, as vectors may move buffers during resize. This fix puts the `DeviceScan` allocation into a `str::shared_ptr`, which 1) allocates the memory on heap and 2) keeps track of the memory through reference counting. Thus it is safe to store the `shared_ptr` across different vectors/maps.
commit 94ab1bb4 (2021-05-30): Use shared_ptr to keep track of DeviceScan across multiple collections

Fixes #681.

Relying on memory allocated by `std::vector` via direct pointers is dangerous,
as vectors may move buffers during resize. This fix puts the `DeviceScan`
allocation into a `str::shared_ptr`, which 1) allocates the memory on heap and
2) keeps track of the memory through reference counting. Thus it is safe to
store the `shared_ptr` across different vectors/maps.
closed 2021-05-30 by f4exb