f4exb/sdrangel #684: DownChannelizer does not have 0dB gain

srcejon, 2020-11-04:
While investigating whether or not there is a benefit to higher sample rates in the ADS-B demod, I noticed that the DownChannelizer does not have 0dB gain, whereas the rational downsampler does. This can be seen in the ChannelAnalyzer:

Here I'm replaying an IQ file at 6MSa/s, and without downsampling, the spectrum is:

[image]

If I enable the rational downsampler, to halve the rate, the average power level remains the same, as I'd expect:

[image]

However, if I try halving the rate with the DownChannelizer instead, then there is a 2-3dB drop:

[image]

Going to 1/4 there's a further 2-3dB drop:

[image]

It should be possible for half-band filters to have 0dB gain, which I think would be preferable, as it should result in better demodulation performance for weak signals.

f4exb, 2020-11-04:
I have to double check but this is normal to see a ~3 dB drop in the noise floor each time the sample rate is halved as there is a mathematical 1 bit gain. Not quite an actual gain but it should effectively lower the noise floor. What wouldn't be right is if a constant signal is also dropped by 3 dB. The signal value should be easier to spot with the scope display.

srcejon, 2020-11-04:
Yep, you're right. I used a more stable signal source and the tone amplitude remains at a stable level, with just the noise reducing. I thought I saw the signal reducing as well earlier, but I can't reproduce it. Sorry about that!

closed 2020-11-04 by srcejon