f4exb/sdrangel #689: SDRangel version 6

f4exb, 2020-11-07:
There has been evolution in the master branch recently that make the desire to keep an ultra stable master branch and fast evolving tip branch (v5) in fact obsolete and not desirable anymore. Moreover quite a few of these changes come from contributors who legitimately work from the master branch.

With this in mind keeping the v5 version range and branch in parallel has become a pain. The v5 branch however has always been in sync with the latest updates and has the more features so it has to become the new master branch while the present master branch will be kept for reference under another name. The dev branch will be dropped in favor of feature branches for each new running development.  

The recent addition of feature plugins would have deserved a new major release but was made a new minor release in each branch because of the split in master and v5. So after this operation versioning will be upgraded to the v6 major versions range.

This will be done after all the current pending active PRs are merged.

According to [this post](https://stackoverflow.com/questions/2763006/make-the-current-git-branch-a-master-branch) this should do the trick:

```sh
# Make sure everything is pushed up to your remote repository (GitHub):
git checkout master
# Overwrite "master" with "v5" branch:
git reset --hard v5
# Force the push to your remote repository:
git push -f origin master
```
It should be noted that any other local copies of the repository will need to `git reset --hard origin/master` next time they want to pull, else git will try to merge the changes into their (now) divergent local.

f4exb, 2020-11-07:
The upcoming pieces of functionality coming from v5 are:
  - MIMO support
  - Websocket spectrum server
  - Additional plugins:
    - ChirpChat (LoRa extension) modulator/demodulator
    - SigMF file support

kasper93, 2020-11-08:
Force pushing to live public repository is quite evil. Have you considered doing merge and resolving conflicts in favor of v5 branch? Which in practice would leave commit history in place of two trees, don't lose changes that might be missing from v5 branch. This is generally "git" way of doing things.

> Moreover quite a few of these changes come from contributors who legitimately work from the master branch.

It is generally rule of thumb to have most recent development HEAD in branch that is set to `default` in github, whatever the name is. So this is way everyone start working form this branch.

f4exb, 2020-11-08:
> Have you considered doing merge and resolving conflicts in favor of v5 branch?

Yes and this is such a large amount of work that I certainly dismiss it. However I am reconsidering the 'evil" operation as you mention it and instead of trying to solve conflicts and get lost with them I am considering importing plugins and rewrite the missing code. For the pieces in the core functionality like websocket spectrum this is just rewriting the code.

> It is generally rule of thumb to have most recent development HEAD in branch that is set to default in github, whatever the name is. So this is way everyone start working form this branch.

Yes this is why swapping branches would also be a possibility but needs co-ordination which is hard and in fact impossible to obtain when the code is out in the nature and this leads us back to the "evil" thing.

I will still try the `git merge` command and see what it leads to.

f4exb, 2020-11-09:
Websocket spectrum server will be handled differently and implemented later.

f4exb, 2020-11-12:
Make sure https://github.com/f4exb/sdrangel/tree/v5/scriptsapi is upgraded as well

f4exb, 2020-11-12:
Deployed in v6.0.0

closed 2020-11-12 by f4exb