f4exb/sdrangel #700: git tag with $ in it causes build failure with ninja

srcejon, 2020-11-10:
After pulling latest git master, when building with ninja, I get:

ninja: error: build.ninja:65: bad $-escape (literal $ must be written as $$)

Which is refering to:

-DSDRANGEL_VERSION=\"4.21.3sdrangel-v$(appveyor_build_version)-19-gdc4075532\"

This version string appears to be generated by sdrangel/CMakeLists.txt:

git describe --tags

Which returns:

sdrangel-v$(appveyor_build_version)-19-gdc4075532

f4exb, 2020-11-10:
Tag deleted

srcejon, 2020-11-10:
Thanks. Even after fetching/pulling and 'git fetch --tags --all' I was still getting the same response from git describe --tags. It seems it needs to be deleted locally as well, with:

git tag -d sdrangel-v$(appveyor_build_version)

f4exb, 2020-11-10:
I tried this locally but no matter what it things that $(appveyor_build_version) is a command. Eventually I had to remove it from `.git/ref/tags` simply with `rm` command

srcejon, 2020-11-10:
That was the command I used in a cmd.exe shell which has different variable expansion. For Linux, it's:

git tag -d sdrangel-v\$\(appveyor_build_version\)

f4exb, 2020-11-13:
There could be a glitch in `appveyor.yml` when processing tags and auto deployment in Github releases. It could be nice to get this working however I would normally run this manually from the `master` version.

f4exb, 2020-11-21:
I am going to drop this one

closed 2020-11-21 by f4exb