f4exb/sdrangel #702: help with compile

hmne, 2020-11-10:
hi there i want some help withe sdrangel  compile stage cmake 
i want to add CM265cc , MBElib , SerialDV , DSDcc , Codec2/FreeDV , LibSigMF and XTRX

all the steps ok but for sdrangel not ok i get this cmake waring
```

CMake Warning:
  Manually-specified variables were not used by the project:

    CM256CC_DIR
    CODEC2_DIR
    DSDCC_DIR
    LIBSIGMF_DIR
    MBE_DIR
    SERIALDV_DIR
    XTRX_DIR
/opt/install/

-- Build files have been written to:
```

ps i used the wiki compile steps all ok 
[https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux](https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux)

and this command 

`cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON -DXTRX_DIR=/opt/install/xtrx-images/include/ -DCM256CC_DIR=/opt/install/cm256cc/include/ -DDSDCC_DIR=/opt/install/dsdcc/include/ -DSERIALDV_DIR=/opt/install/serialdv/include/ -DMBE_DIR=/opt/install/mbelib/include/ -DCODEC2_DIR=/opt/install/codec2/include/ -DLIBSIGMF_DIR=/opt/install/libsigmf/include/ -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..`

the folders on `/opt/install/`

`codec2  dsdcc  libsigmf  mbelib  serialdv  xtrx-images`

f4exb, 2020-11-10:
Please read [the instructions](https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux) thoroughly. In particular:
```sh
cd /opt/build
git clone https://github.com/f4exb/sdrangel.git
cd sdrangel
mkdir build; cd build
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DMIRISDR_DIR=/opt/install/libmirisdr \
-DAIRSPY_DIR=/opt/install/libairspy \
-DAIRSPYHF_DIR=/opt/install/libairspyhf \
-DBLADERF_DIR=/opt/install/libbladeRF \
-DHACKRF_DIR=/opt/install/libhackrf \
-DRTLSDR_DIR=/opt/install/librtlsdr \
-DLIMESUITE_DIR=/opt/install/LimeSuite \
-DIIO_DIR=/opt/install/libiio \
-DPERSEUS_DIR=/opt/install/libperseus \
-DXTRX_DIR=/opt/install/xtrx-images \
-DSOAPYSDR_DIR=/opt/install/SoapySDR \
-DUHD_DIR=/opt/install/uhd \
-DCM256CC_DIR=/opt/install/cm256cc \
-DDSDCC_DIR=/opt/install/dsdcc \
-DSERIALDV_DIR=/opt/install/serialdv \
-DMBE_DIR=/opt/install/mbelib \
-DCODEC2_DIR=/opt/install/codec2 \
-DLIBSIGMF_DIR=/opt/install/libsigmf \
-DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
make -j4 install
```
Do you see the difference with yours?

hmne, 2020-11-10:
nope same thing 
copy past your command and nothing 
```
CMake Warning:
  Manually-specified variables were not used by the project:

    AIRSPYHF_DIR
    AIRSPY_DIR
    BLADERF_DIR
    CM256CC_DIR
    CODEC2_DIR
    DSDCC_DIR
    HACKRF_DIR
    IIO_DIR
    LIBSIGMF_DIR
    LIMESUITE_DIR
    MBE_DIR
    MIRISDR_DIR
    PERSEUS_DIR
    RTLSDR_DIR
    SERIALDV_DIR
    SOAPYSDR_DIR
    UHD_DIR
    XTRX_DIR

-- Build files have been written to: /opt/build/sdrangel/build

```

my command 
```

  Manually-specified variables were not used by the project:

    CM256CC_DIR
    CODEC2_DIR
    DSDCC_DIR
    LIBSIGMF_DIR
    MBE_DIR
    SERIALDV_DIR
    XTRX_DIR

-- Build files have been written to: /opt/build/sdrangel/build

```
the folders there and all files one there place cheeked 10 times 

its only find things on  `/usr/local/include/`  automatically  but if i copy the foldres to the /uder/loc/include/ 
and same thing

f4exb, 2020-11-10:
I don't know I have followed this procedure a zillion of times and it works.

closed 2020-11-10 by f4exb

hmne, 2020-11-10:
i re installed new OS only for this to work this is the first thing installed in my system after update but its not working all dependency are ok 
and the folders ok to by like this ?      `/opt/install/codec2/include/codec2`
ps my os kali linux 5.9.0    

**`I don't know I have followed this procedure a zillion of times and it works.`**

there's first time for everything

f4exb, 2020-11-10:
Sorry the issues you are facing show you don't know how to compile software. I cannot spend time on the basics.

hmne, 2020-11-10:
`Sorry the issues you are facing show you don't know how to compile software. I cannot spend time on the basics. I suggest you use a package or simply not use this software`

yes theat was a mistake on my part in writing because i did not sleep for every , sorrrry