f4exb/sdrangel #707: SDRAngle install error

7ue4ssA, 2020-11-14:
I have installed all plugins I needed and all dependencies, but in the last step the " make -j4 install " everythings is working fine and without problems until to
 ****Scanning dependencies of target remotesinksrv
[ 38%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinksrv_autogen/mocs_compilation.cpp.o****
From there on up to %39 I get finally an error prompt.
**[ 39%] Built target remotesinksrv
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2**

Scanning dependencies of target remotesinksrv
[ 38%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinksrv_autogen/mocs_compilation.cpp.o
/opt/build/sdrangel/sdrgui/gui/glspectrum.cpp: In member function ‘void GLSpectrum::applyChanges()’:
/opt/build/sdrangel/sdrgui/gui/glspectrum.cpp:1254:13: error: ‘class QFontMetrics’ has no member named ‘horizontalAdvance’
  int M = fm.horizontalAdvance("-");
             ^~~~~~~~~~~~~~~~~
/opt/build/sdrangel/sdrgui/gui/glspectrum.cpp:1264:21: error: ‘class QFontMetrics’ has no member named ‘horizontalAdvance’
  m_rightMargin = fm.horizontalAdvance("000");
                     ^~~~~~~~~~~~~~~~~
/opt/build/sdrangel/sdrgui/gui/glspectrum.cpp:1803:20: error: ‘class QFontMetrics’ has no member named ‘horizontalAdvance’
       shift = - fm.horizontalAdvance(ftext);
                    ^~~~~~~~~~~~~~~~~
[ 38%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesink.cpp.o
sdrgui/CMakeFiles/sdrgui.dir/build.make:808: recipe for target 'sdrgui/CMakeFiles/sdrgui.dir/gui/glspectrum.cpp.o' failed
make[2]: *** [sdrgui/CMakeFiles/sdrgui.dir/gui/glspectrum.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 38%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinkbaseband.cpp.o
/opt/build/sdrangel/sdrgui/gui/glscope.cpp: In member function ‘void GLScope::setUniqueDisplays()’:
/opt/build/sdrangel/sdrgui/gui/glscope.cpp:1126:16: error: ‘class QFontMetrics’ has no member named ‘horizontalAdvance’
     int M = fm.horizontalAdvance("-");
                ^~~~~~~~~~~~~~~~~
/opt/build/sdrangel/sdrgui/gui/glscope.cpp: In member function ‘void GLScope::setVerticalDisplays()’:
/opt/build/sdrangel/sdrgui/gui/glscope.cpp:1310:16: error: ‘class QFontMetrics’ has no member named ‘horizontalAdvance’
     int M = fm.horizontalAdvance("-");
                ^~~~~~~~~~~~~~~~~
/opt/build/sdrangel/sdrgui/gui/glscope.cpp: In member function ‘void GLScope::setHorizontalDisplays()’:
/opt/build/sdrangel/sdrgui/gui/glscope.cpp:1493:16: error: ‘class QFontMetrics’ has no member named ‘horizontalAdvance’
     int M = fm.horizontalAdvance("-");
                ^~~~~~~~~~~~~~~~~
/opt/build/sdrangel/sdrgui/gui/glscope.cpp: In member function ‘void GLScope::setPolarDisplays()’:
/opt/build/sdrangel/sdrgui/gui/glscope.cpp:1675:16: error: ‘class QFontMetrics’ has no member named ‘horizontalAdvance’
     int M = fm.horizontalAdvance("-");
                ^~~~~~~~~~~~~~~~~
[ 38%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinksink.cpp.o
sdrgui/CMakeFiles/sdrgui.dir/build.make:730: recipe for target 'sdrgui/CMakeFiles/sdrgui.dir/gui/glscope.cpp.o' failed
make[2]: *** [sdrgui/CMakeFiles/sdrgui.dir/gui/glscope.cpp.o] Error 1
CMakeFiles/Makefile2:3982: recipe for target 'sdrgui/CMakeFiles/sdrgui.dir/all' failed
make[1]: *** [sdrgui/CMakeFiles/sdrgui.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 39%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinksettings.cpp.o
[ 39%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinkwebapiadapter.cpp.o
[ 39%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinksender.cpp.o
[ 39%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinkfifo.cpp.o
[ 39%] Building CXX object pluginssrv/channelrx/remotesink/CMakeFiles/remotesinksrv.dir/remotesinkplugin.cpp.o
[ 39%] Linking CXX shared library ../../../lib/pluginssrv/libremotesinksrv.so
[ 39%] Built target remotesinksrv
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

I would be very very happy if anyone could help me out. What the issue could be.

f4exb, 2020-11-14:
Please read https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux carefully:

> Compiling from source is assumed to be done in a Linux system with an Ubuntu 20.10 distribution. 20.10 is preferred over the 20.04 LTS because it solves 2 nasty bugs:
>
> #615: resolved by Qt 5.14.2
> #600: resolved by kernel 5.8
> Be aware that if you choose a different distribution with earlier versions these bugs may re-appear. Compilation instructions are  not different from 20.04.
>
> The minimal standard required for the C++ compiler is c++14

The issue here is that your version of Qt is too old.

7ue4ssA, 2020-11-14:
Thanks for taking your time and giving me advice. I cant for some reasons not use 20.04 version.
Is here a version of SDRAngel which I can use for Ubuntu 18.04 Version?

f4exb, 2020-11-14:
I think 18.04 is with Qt 5.9.9 which is getting quite old so my advice would still be to upgrade. Note that Qt 5.11.3 is enough since it compiles with Debian Buster using this version. 

I think the issue you have comes from a change to remove a deprecation warning in Qt 5.14.2 and this came with the move to Ubuntu 20.10 for development occurring ~22 Oct. I guess version 4.20.0 (11 Oct) should work so you can checkout this tag.

7ue4ssA, 2020-11-14:
A week ago I installed SDRAngel and it runs without problem on Ubuntu 18.04 on a other Computer. (With the lastest version at one week ago).
But Thanks for your help I will try and see if it's working then.

7ue4ssA, 2020-11-16:
Hi again, I was able to solve the problem. I have installes the previous version of SDRAngel. But yes there is the issue with file recording. 
I have although a Computer with the latest version of SDRAngel and I don't know if there is a problem with the file sink. Because even if I click of record I can not find any recorded file in any direction if my computer.

But the main problem with the installation is solved and it's running without any other problems so far.

f4exb, 2020-11-17:
It will not choose a file name for you. Please refer to: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/filesink/readme.md

closed 2020-12-13 by f4exb