f4exb/sdrangel #708: Hermes Lite 2 support

yen0p, 2020-11-15:
Hello,

I would like to start this message by thanking you for the amazing work done !

I've already used SDRAngel with rx only receiver and it works as expected (even more).
I'm thinking about replacing my FT991A with 1 or more SDR with TX capabilities. 

Hermes Lite 2 seems to be a good candidate, as it's not so expensive and have quite good specifications. HL2 is also  fully open source which is something i really appreciate.
Based on this link : https://github.com/f4exb/sdrangel/wiki/Sample-source-plugins-(Rx-devices) i don't think HL2 is currently supported, is it in the roadmap ?

Best regards
F4INE, Nicolas

f4exb, 2020-11-15:
Hi,

Hermes/Metis is supported but as MISO (Multiple In Single Out) device: https://github.com/f4exb/sdrangel/wiki/Device-MIMO-plugins-(Multiple-In-Multiple-Out)#metis-hpsdr-miso-plugin However it was used and tested only with Red Pitaya wih Pavel Demin's firmware.

yen0p, 2020-11-15:
Thank you for the information, I ll try to find a HL2 owner and make him/her test.

closed 2021-09-09 by github-actions[bot]

heathhey, 2022-08-08:
I now have two HL2s and would be willing to test. The metis connector works (it recieves packets and can demodulate SSB), but there isn't anything to control the various filters on the n2adr board, nor can I see other controls such as LNA setting.

gm8arv, 2023-01-23:
Did you get any further with your tests?  I am expecting an HL2 within a couple of months so this is of interest to me.  Thanks!  David