f4exb/sdrangel #711: Remote input source / output sink

srcejon, 2020-11-16:
While doing some testing, I wanted to feed the output of the AM Modulator in to an AM Demodulator within the same instance of SDRangel. So I tried to do this by using the Remote Output sink and Remote Input source. 

Firstly, using the default settings, I notice the received spectrum has a much lower level (-40dB) than what is transmitted. Is this to be expected? I would have expected a similar level.

[image]

[image]

Secondly, I seem to get a large amount of dropped packets, even at low sample rates. For example, I originally started on Windows, where it would be unable to stream at 48kSa/s without error:

[image]

On the same (fast) PC, using Linux, 48kSa/s seems to work OK, but 100kSa/s wouldn't. That seems surprisingly low for local loopback. Increasing FEC blocks didn't solve the problem. Performance appears to be about the same if the connection is between two separate instances of SDRangel on the same PC.

Any thoughts on what I should expect or might need to investigate? Thanks.

f4exb, 2020-11-16:
This is not the intention behind the "remote" plugins. What they are made for:
  - Rx path: Input device -> Remote sink -> network -> Remote input -> Channel(s) 
  - Tx path: Channel(s) -> Remote output -> network -> Remote source -> Output device

I run 500 kSa/s from a RPi to the laptop over the LAN without major issues until I load the laptop big time with a big compilation (make -j12 of sdrangel for example... - running on a 6 core with HT).

f4exb, 2020-11-21:
The limitation on sample rate could be a consequence of "auto throttling" that you see here: https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/remoteinput/remoteinputudphandler.cpp#L302 . The `tick` method is called every 50ms and with increasing sample rate comes an increasing amount of data for which this mechanism may be not designed for. There is a limit anyway on what timers can handle so reducing the `tick` period will encounter a limitation inevitably.

This "auto throttling" was found necessary because UDP packets do not arrive regularly and they do more often come as bursts so the purpose is to make the flow smoother and also to keep the balance of the FIFO to compensate for clocks skew between different systems. You have a representation of this balance in the GUI with the gauge below the DC/IQ buttons. This obviously can work only at relatively slow rates in practice it will work fine at the very common audio sample rate of 48 kSa/s and may be a bit higher but not very much larger.

srcejon, 2020-11-22:
With regards to the -40dB loss, I believe this is due to TX being 16-bit and RX being 24-bit. If I build SDRangel with RX=16-bit, then this loss does not occur. Could presumably be fixed with some scaling, but as you said, this isn't the intended use case. 

(Slight side issue: SDRangel failed to build with 16-bit RX on Windows: dspdevicemimoengine.cpp(459): error C2398: Element '1': conversion from 'int' to 'FixReal' requires a narrowing conversion).

I then switched to Remote Sink to Remote input, as suggested, and there, I observed similar packet loss. (i.e. it was working on Linux at around 62.5kSa/s, but packets were lost at 125kSa/s.) 

Haven't had a chance to look in to any code yet - but this is on an i9 7990XE so should be able to match a Pi. I'll run a few more tests when I get a chance.

f4exb, 2020-11-24:
16-bit RX (error) on Windows and (warning) in Linux/gcc should be addressed by this commit: https://github.com/f4exb/sdrangel/commit/ce2aad5a7a35c320152404f135801fe0ff947326

closed 2021-09-09 by github-actions[bot]