f4exb/sdrangel #713: Rest API: UDPSinkSettings key mismapped

dcampos2015, 2020-11-17:
Hello,
I'm playing around with the REST API for put / patching settings for the UDPSink channel using information retrieved from the swagger docs.
I have two channels, respectively 0 and 1, both set up as UDPSink and want to update the configuration on sink 1. 

From swagger, I get the following curl command (after removing unneeded  config options: 
`curl -X PATCH "http://localhost:8091/sdrangel/deviceset/0/channel/1/settings" -H  "accept: application/json" -H  "Content-Type: application/json" -d "{\"UDPSinkSettings\": {\"inputFrequencyOffset\": 287600},\"channelType\": \"UDPSink\",\"direction\": 0}"`

but this fails with error 400 / invalid JSON request:
[image]

Poking around, I found that m_channelTypeToSettingsKey maps UDPSink to UDPSourceSettings and UDPSource to UDPSinkSettings: https://github.com/f4exb/sdrangel/blob/fbe77a1922564d6df770522a6a7f434d161fdfa4/sdrbase/webapi/webapiutils.cpp#L141

Interestingly enough, specifying UDPSourceSettings instead causes a segmentation fault:

curl request: `curl -X PATCH "http://localhost:8091/sdrangel/deviceset/0/channel/1/settings" -H  "accept: application/json" -H  "Content-Type: application/json" -d "{\"UDPSourceSettings\": {\"inputFrequencyOffset\": 287600},\"channelType\": \"UDPSink\",\"direction\": 0}"
curl: (52) Empty reply from server
`

 and associated output:
```
2020-11-17 18:23:15.852 (D) GLShaderTextured::initializeGL: m_useImmutableStorage:  true
2020-11-17 18:23:17.807 (D) HttpConnectionHandler (0x5643d1b20be0): constructed
2020-11-17 18:23:17.807 (D) HttpConnectionHandler (0x5643d1b20be0): received request from 127.0.0.1 (PATCH) /sdrangel/deviceset/0/channel/1/settings
./start.sh: line 4: 266944 Segmentation fault      (core dumped) ./usr/bin/sdrangel

```

Remapping UDPSink and UDPSource to their matching Settings keys + recompiling fixes this and allows me to submit API requests normally. With this in mind, am I correct in suspecting that these keys are indeed mapped incorrectly and this is (from my very quick overview) the only change needed?

Let me know if there's any additional information I can provide. Thanks

f4exb, 2020-11-18:
That's probably the case as at some point the "source" and "sink" denomination was swapped to match the other plugins convention.  In the process the REST API changes may have been overlooked.

f4exb, 2020-11-18:
Confirmed

f4exb, 2020-11-22:
Implemented in v6.1.0

closed 2020-11-22 by f4exb