f4exb/sdrangel #716: Core dump at startup

f4exb, 2020-11-20:
Compiled fresh from master. Then:
```
2020-11-20 21:53:06.606 (D) MainWindow::MainWindow: load plugins...                                                                                                                                                                         
2020-11-20 21:53:06.606 (D) PluginManager::loadPluginsPart folder /opt/build/sdrangel/build/../lib/sdrangel/plugins is empty                                                                                                                
2020-11-20 21:53:06.607 (D) PluginManager::loadPluginsDir: fileName: libbeamsteeringcwmod.so                                                                                                                                                
Segmentation fault (core dumped)   
```

f4exb, 2020-11-20:
```
[Current thread is 1 (Thread 0x7f9830d3f000 (LWP 1272716))]
(gdb) bt
#0  std::__atomic_base<int>::load (__m=std::memory_order_relaxed, this=0x0) at /usr/include/c++/10/bits/atomic_base.h:426
#1  QAtomicOps<int>::loadRelaxed<int> (_q_value=...) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qatomic_cxx11.h:239
#2  0x00007f982c019af2 in QBasicAtomicInteger<int>::loadRelaxed (this=0x0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qbasicatomic.h:107
#3  0x00007f982c01a74c in QtPrivate::RefCount::ref (this=0x0) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qrefcount.h:55
#4  0x00007f982c01a9f7 in QString::QString (this=<optimized out>, other=..., this=<optimized out>, other=...) at /usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:1051
#5  0x00007f982c02320c in __static_initialization_and_destruction_0 (__initialize_p=1, __priority=65535) at /opt/build/sdrangel/plugins/channelmimo/beamsteeringcwmod/beamsteeringcwmodplugin.cpp:37
#6  0x00007f982c023343 in _GLOBAL__sub_I_beamsteeringcwmodplugin.cpp(void) () at /opt/build/sdrangel/plugins/channelmimo/beamsteeringcwmod/beamsteeringcwmodplugin.cpp:93
#7  0x00007f982c027eee in _sub_I_65535_0.0 () at /opt/build/sdrangel/sdrbase/dsp/channelmarker.h:28
#8  0x00007f98372f4dbe in ?? () from /lib64/ld-linux-x86-64.so.2
#9  0x00007f98372f4ea8 in ?? () from /lib64/ld-linux-x86-64.so.2
#10 0x00007f9835141505 in __GI__dl_catch_exception (exception=<optimized out>, operate=<optimized out>, args=<optimized out>) at dl-error-skeleton.c:182
#11 0x00007f98372f9505 in ?? () from /lib64/ld-linux-x86-64.so.2
#12 0x00007f98351414a8 in __GI__dl_catch_exception (exception=0x7ffd88648070, operate=0x7f98372f9150, args=0x7ffd88648090) at dl-error-skeleton.c:208
#13 0x00007f98372f8b5d in ?? () from /lib64/ld-linux-x86-64.so.2
#14 0x00007f983382734c in dlopen_doit (a=a@entry=0x7ffd886482c0) at dlopen.c:66
#15 0x00007f98351414a8 in __GI__dl_catch_exception (exception=exception@entry=0x7ffd88648260, operate=operate@entry=0x7f98338272f0 <dlopen_doit>, args=args@entry=0x7ffd886482c0) at dl-error-skeleton.c:208
#16 0x00007f9835141573 in __GI__dl_catch_error (objname=objname@entry=0x5633135122e0, errstring=errstring@entry=0x5633135122e8, mallocedp=mallocedp@entry=0x5633135122d8, operate=operate@entry=0x7f98338272f0 <dlopen_doit>, 
    args=args@entry=0x7ffd886482c0) at dl-error-skeleton.c:227
#17 0x00007f9833827b89 in _dlerror_run (operate=operate@entry=0x7f98338272f0 <dlopen_doit>, args=args@entry=0x7ffd886482c0) at dlerror.c:170
#18 0x00007f98338273d8 in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#19 0x00007f9835673d1f in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007f983566d58d in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007f983566db87 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#22 0x00007f983565eae6 in QPluginLoader::load() () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#23 0x00007f9836842e52 in PluginManager::loadPluginsDir (this=0x56331363bfb0, dir=...) at /opt/build/sdrangel/sdrbase/plugin/pluginmanager.cpp:218
#24 0x00007f9836841b61 in PluginManager::loadPluginsPart (this=0x56331363bfb0, pluginsSubDir=...) at /opt/build/sdrangel/sdrbase/plugin/pluginmanager.cpp:99
#25 0x00007f983684189b in PluginManager::loadPlugins (this=0x56331363bfb0, pluginsSubDir=...) at /opt/build/sdrangel/sdrbase/plugin/pluginmanager.cpp:63
#26 0x00007f98370c4ab6 in MainWindow::MainWindow (this=<optimized out>, logger=<optimized out>, parser=..., parent=<optimized out>, this=<optimized out>, logger=<optimized out>, parser=..., parent=<optimized out>)
    at /opt/build/sdrangel/sdrgui/mainwindow.cpp:191
#27 0x0000563311d6c489 in runQtApplication (argc=5, argv=0x7ffd88648bf8, logger=0x563313557070) at /opt/build/sdrangel/app/main.cpp:127
#28 0x0000563311d6c74c in main (argc=5, argv=0x7ffd88648bf8) at /opt/build/sdrangel/app/main.cpp:137
```

f4exb, 2020-11-20:
Falling back #715 and #714 solves the issue.

f4exb, 2020-11-21:
I think issue is with `CMAKE_INTERPROCEDURAL_OPTIMIZATION`. This is known to cause trouble. I think this is why the link of plugins is broken.

kasper93, 2020-11-21:
Yes it is the cause, but the underlying root cause is not known yet. There is no reason for LTO (which this flag enables) to not work really. But as we can see  some incompatibilities/issues arise. FWIW it works compiled with clang and MSVC (LTCG is another beast, but we had it enabled before my changes). 

I will debug what is really going on there and get back to you.

f4exb, 2020-11-21:
I was thinking this was incompatible with the Qt plugins because it would try to optimize over the whole compilation unit but a plugin should be considered as an isolated one. However if this is GCC specific it could be the combination of both. This is just an idea. Now I think I came across this situation a long time ago and eventually gave up on this optimization.

f4exb, 2020-11-22:
Released in v6.1.0

closed 2020-11-22 by f4exb