f4exb/sdrangel #718: Hackrf one can not run!

longyucheng123, 2020-11-22:
Hello! Hackrf one can not run in the latest version of sdrange 6.0.2-win64 software. After startup, the program will flash back, hoping to be updated and solved.

kasper93, 2020-11-22:
I don't have hackrf to test. Does sdrangel start when device is disconnected? Could you test older version and tell which one is last working?

longyucheng123, 2020-11-23:
Hello! Please see the attached figure, select this to run the program to exit, select soapysdr hackrf is normal, hope to update the software to solve, thank you!发自我的华为手机-------- 原始邮件 --------发件人： Kacper Michajłow <notifications@github.com>日期： 2020年11月22日周日 22:40收件人： f4exb/sdrangel <sdrangel@noreply.github.com>抄送： longyucheng123 <longyucheng82@126.com>, Author <author@noreply.github.com>主    题： Re: [f4exb/sdrangel] Hackrf one can not run! (#718)
I don't have hackrf to test. Does sdrangel start when device is disconnected? Could you test older version and tell which one is last working?

reischle, 2020-11-23:
I confirm the problem. SDRangel starts up, if the HackRF is not connected, or if previously a different device had been selected. If the hackrf is then selected from the device list, it also throws an unhandled exception:
"Unhandled exception at 0x00007FF85AED0130 (libusb-1.0.dll) in sdrangel.exe: 0xC0000005: Access violation reading aztposition 0x0000000000000220"

Ellude1, 2020-12-04:
Similar situation here on Windows 10. I have updated sdrangel from 4.11.9 to 6.2.0. Everything worked fine with the older version, however for the newer version it was at first stuck at load input files if hackrf was connected failing to launch. With hackrf being disconnected program launched correctly and afterwards the program launches normally every time even with hackrf connected. However selecting hackrf as a source closes the program after a few seconds without an error. Anywhere where I could extract logs? I also have almost no idea what I'm doing with sdr radio so it might be something unrelated, however sdr# and other software seem to work fine. Have a nice day!

faridfarhan28, 2020-12-09:
hi, i have same problem. HackRF not running on windows when i plugged HackRF to USB. but i try to unplug hackRF and after that i try running SDRangel and works well. but, when i plug to USB, HackRF was not detected (picture on below) meanwhile i use other SDR (Nooelec, RTL-SDRv3) is detected and work well normaly. I try to downgrade SDRangel to 4.11.9 and work normally on hackRF. I hope you can fix it.

ceers.... de YC1HVZ :) 
 
HackRF plugged
[image]

nooelec nesdr smartXTR plugged
[image]

kasper93, 2020-12-11:
Could you follow this https://www.meziantou.net/tip-automatically-create-a-crash-dump-file-on-error.htm and upload minidump created?

RaspCla, 2020-12-11:
Hi,
I started to use this cool tool yesterday and I love it.
Tested with SDR-RTL Dongle and works fine
Tested with HackRF, it works but I found some issues (i will start a new issue)

I use Windows 10 (latest version 20H2), WINUSB (V10.0.1904.1) as driver for HackRF and the SoapySDR.... module inside SDRangel, see attachments
[image]
[image]

faridfarhan28, 2020-12-14:
hi mr.Edouard RaspCla. thanks you now HackRF is working on 6.3.3 by SoapySDR. Thank you 👍 

[image]

RaspCla, 2020-12-14:
yes, but SDR Angel crashes if I select HackRF[0]... directly

faridfarhan28, 2020-12-15:
> yes, but SDR Angel crashes if I select HackRF[0]... directly

i thing don't use direct HackRF[0] because SDRAngel has SoapySDR plugin driver so it's support correctly

Drackonic, 2020-12-19:
Having the issue as well on Windows 10 version 2004, seems windows just doesn't want to play nice with the newer versions of this currently.

Luckily the soapysdr workaround currently is working well enough for me even if it's a bit more awkward to control the radio with it.

Drackonic, 2020-12-20:
In WinDgb It produces this when you attempt to load the normal HackRF

``(1264.e90): Access violation - code c0000005 (first chance) First chance exceptions are reported before any exception handling. This exception may be expected and handled. libusb_1_0!libusb_interrupt_transfer+0x18b0: 00007ffe`0b1f0130 488b8120020000 mov rax,qword ptr [rcx+220h] ds:00000000`00000220=????????????????``

``(1264.e90): Access violation - code c0000005 (!!! second chance !!!) libusb_1_0!libusb_interrupt_transfer+0x18b0: 00007ffe`0b1f0130 488b8120020000 mov rax,qword ptr [rcx+220h] ds:00000000`00000220=????????????????``

Hopefully this is of any help at all.

kasper93, 2020-12-20:
Backtrace or minidump will help as stated here https://github.com/f4exb/sdrangel/issues/718#issuecomment-743192899 I don't have hackrf so need more info if you want this fixed.

Drackonic, 2020-12-20:
[Sdr angel dump.zip](https://github.com/f4exb/sdrangel/files/5720651/Sdr.angel.dump.zip)

Will this suffice?

raplin, 2021-01-17:
Same issue for me (Win7; hackrf, works on SDRAngel before about V6).   It's a null pointer ref in libusb, Sorry I don't have the dev environment set up to find root cause.. 
[image]

f1gpa, 2021-01-31:
Hello, problem with me also, connected portapack and last version sdrangel under Windows 10. Sdrangel closes as soon as the click on hackrf. I uninstalled, cleaned with ccleaner but nothing helps. Identical on a second PC! Cordially. F1GPA

4JAM4, 2021-02-14:
Also confirming all versions of SDRAngel 6.5.* with Windows 10 (OS Build 19041.746) crash with HackRF One plugged in, and will start without HackRF One not plugged in (but then HackRF One not found).  

I have tested this with two HackRF One units.  Both HACKRFOne devices work fine with SDRAngel-4.14.10-win64.

Happy to spend the time to determine when (ie what version) the HackRF One option stopped working if the sdrangel-x.y.z-win64.exe files can be made available for testing (I note that they are no longer on github).

TGCFabian, 2021-02-18:
@4JAM4 I've noticed the support broke after the implementation of SoapySDR Within SDRAngel(Somewhere after 4.21.3)
Personally ive tried it on several windows machines of mine, and had no luck, unless i would use SoapySDR Library
Every single time it would crash, the console output would show "Segmentation Fault"

4JAM4, 2021-02-21:
https://github.com/TGCFabian  that seems to be about right.  I am currently using the work around posted by 
https://github.com/faridfarhan28   and also have HackRF is working on windows. 6.3.3 by SoapySDR.
I will run with this on WIndows. The issue is only specific to Windows as all appears OK on the Ubuntu O/S environment with an Rpi-4 .

TGCFabian, 2021-02-21:
@4JAM4 Using the SoapySDR Implementation is indeed the way to use for now, untill the previous way is fixed :)
Hope it gets fixed soon :D

Smilerification, 2021-11-12:
Any updates on this issue?

TGCFabian, 2021-11-12:
Sup @Smilerification!
 for what i know, there hasent been any update, but people are able to use SoapySDR within SDRAngel, this does work as stated above, but id still love seeing the old one be fixed
 
 SoapySDR is a multi SDR Driver, compared to the normal stand alone direct one, 
 Soapy does fully support the HackRF (Both RX and TX), so if you'd have to use the hackrf with SDRAngel, give it a try
 
 Thats atleast what im doing for now :3

Smilerification, 2021-11-13:
Yea, have been fiddling around with SoapySDR, but the gain controls drive me nuts... They jump around when you adjust one, making fine tuning near impossible. Also the gain settings are forgotten when you quit SDRangel, forcing me to go through this process over and over.

Will be using SDRangel 4.11.6 from now on, kinda sad I will be missing out on the DATV modulator though. Really, really hope native support for the HackRF One will be fixed at some point.

TGCFabian, 2021-11-13:
Oh right, yea i forgot about the janky gain settings, ive stopped using SDRAngel with my hackrf due to that stuff, 

And yeah, it might be smart to use the older one untill its fixed, i hope it gets fixed soon, that way i can start using it more daily ;D

Smilerification, 2021-11-13:
I think I found a fix. Removing SoapySDR.dll from the installation directory suddenly made the HackRF work. No idea how, but it works for me.

TGCFabian, 2021-11-13:
Okay wow that actually works 😆 😆 😆 😆 

Thanks @Smilerification! Thats kinda a stupid fix but im glad it works
Thanks mate!

Smilerification, 2021-11-13:
NP bro it was thanks to your suspicions of Soapy breaking HackRF.

TGCFabian, 2021-11-13:
haha, i never even thought of deleting Soapy from the installation, weird how the standalone hackrf crashed because of soapy being installed tho, maybe one of those cases where soapy 'claimed' the sdr? not sure, but hey we got it for now :D

Drackonic, 2021-11-13:
Very weird, deleting SoapySDR.dll made everything work exactly as it should. Many thanks to @Smilerification for the workaround.

RaspCla, 2021-11-17:
Thanks a lot for this information. Deleting SoapySDR.dll does also work for my setup

f4exb, 2021-12-17:
Works also for me. This questions whether SoapySDR should be supported at all.

srcejon, 2022-02-03:
As per the mailing list, looks like just removing the HackRF Soapy DLL might do it: 

https://github.com/f4exb/sdrangel-windows-libraries/pull/11

TGCFabian, 2022-02-03:
This fix does indeed work! Good catch srcejon!

Loading times of SDRAngel did go up again, as SoapySDR Checks every single audio device, 

The fix proposed by srcejon should be merged imo, as this will return good functionality to the original plugin

daviderud, 2022-04-07:
Hi, deletion of dll file confirmed working also for me.
To be done also after installing new version of the program.

f4exb, 2022-04-16:
It looks like that for this reason and some others Soapy creates more issues than it is supposed to solve. Instead of removing it completely maybe it should be activated only upon a flag on the command line. TBD with v7.

f4exb, 2022-05-02:
Implemented in v7.0.0-alpha.3

commit 89196c40 (2022-05-01): Massive UI revamping (v7): Make soapy SDR support optional. Fixes #718
closed 2022-05-02 by f4exb