f4exb/sdrangel #720: HAS_NEON flag removed

f4exb, 2020-11-23:
Therefore the following plugins are not compiled anymore on ARM:
  - RemoteSink
  - FileSource (here the flag is useless and should be removed from Cmake file)
  - RemoteSource
  - RemoteOutput
  - RemoteInput

This seriously impairs ARM deployment where remote plugins are essential for remote connection of a Raspberry Pi.

f4exb, 2020-11-23:
This section can probably be removed in all mentioned plugins Cmake files:
```cmake
if (HAS_SSSE3)
    message(STATUS "RemoteSink: use SSSE3 SIMD" )
elseif (HAS_NEON)
    message(STATUS "RemoteSink: use Neon SIMD" )
else()
    message(STATUS "RemoteSink: Unsupported architecture")
    return()
endif()
```

f4exb, 2020-11-23:
Still missing `-DUSE_NEON` to compile the RemoteXXX plugins on ARM64. This flag is expected by `gf256.h` in the `cm256cc` package.

kasper93, 2020-11-23:
It is defined as `-DUSE_ARM_NEON` and `HAS_ARM_NEON` now to match preprocessed macro name. I didn't find any uses of `USE_NEON`, but my editor doesn't search build dir, where external projects are. Sorry about that. This flag could be passes in `ExternalProject_Add` for cm256cc to make sure both are built the same.

f4exb, 2020-11-23:
This would require a change in cm256cc to match the same tags which is of course possible but for the time being I'll keep it like that.

f4exb, 2020-11-23:
Should not this: https://github.com/f4exb/sdrangel/blob/master/cmake/Modules/DetectArchitecture.cmake#L200
be this:
```cmake
detect_extensions(ARM_NEON HAS_ARM_NEON)
```
?

kasper93, 2020-11-23:
No. The second parameter is used to skip detection, for example.

```
detect_extensions(AVX512F)
detect_extensions(AVX2 HAS_AVX512F)
```

Detection of  `AVX512F` will always run, and when it is succeed i.e. defined `HAS_AVX512F` to `true`. We know for sure that AVX2 is also supported, hence we use to information in next one. If AVX512 was not found it will do full detection for AVX2. And similar for the rest. 

It is a shortcut to make detection faster

f4exb, 2020-11-23:
OK got it but it turns out that the compiler flags are a bit of a mess on RPi4 with aarch64 (armv8) kernel. Firstly the extension flag to test is `__ARM_NEON` and not `__ARM_NEON__` which seems to have become obsolete. However trying this flag format on the RPi4 the extension is not picked up although I have:
```bash
g++ -march=native -dM -E - </dev/null | grep NEON
#define __ARM_NEON 1
g++ --version
g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0        
```
Well... since ARM64 always has NEON we may just as well force the flag if architecture is ARM64.

kasper93, 2020-11-23:
> Firstly the extension flag to test is __ARM_NEON and not __ARM_NEON__ which seems to have become obsolete.

Ugh, I have my RPi3 with aarch64 kernel, but with 32-bit userland Raspbian. I tested it for flags before and manually requesting neon. 

```
$ g++ -mfpu=neon -dM -E - </dev/null | grep NEON
#define __ARM_NEON_FP 4
#define __ARM_NEON__ 1
#define __ARM_NEON 1 
$ g++ --version
g++ (Raspbian 8.3.0-6+rpi1) 8.3.0
```

I see that this is problem, so 
Replace
``` Cmake
detect_extensions(ARM_NEON)
```
with
``` Cmake
check_symbol_exists("__ARM_NEON" "" HAS_NEON)
if (HAS_NEON)
  add_compile_definitions(USE_NEON)
endif()
```

This should work in both cases and define old `USE_NEON`

f4exb, 2020-11-23:
Yes it works: https://github.com/f4exb/sdrangel/commit/4059a0be27731b519de9c9c3f0c2bc3fd86c9ca7

f4exb, 2020-11-26:
Released in v6.2.0

closed 2020-11-26 by f4exb