f4exb/sdrangel #721: v. 6.1 does not run on Lubuntu 20.04

loewal, 2020-11-23:
As mentioned. I installed the .deb-file. "Illegal instruction".
Went back to 6.0.2: all fine.......

f4exb, 2020-11-23:
What is the output of `cat /proc/cpuinfo | grep flags | head -n 1` on your machine ?

loewal, 2020-11-23:
This:
cat /proc/cpuinfo | grep flags | head -n 1
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts rep_good nopl cpuid aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm lahf_lm pti tpr_shadow vnmi flexpriority dtherm ida

loewal, 2020-11-23:
I'm running SDRangel on a Dell Vostrro 1700. Old machine indeed.
Is there a way around? I didn't install SDRangel; it came on Dragonos Focal preinstalled.

loewal, 2020-11-23:
B.t.w., on my Windows 10 on my first hd (2 installed), SDRangel runs fine.
Is a do-release-upgrade useful on my Lubuntu?

f4exb, 2020-11-23:
Yes Nehalem expects `sse4_1 sse4_2` extension that I do not see in the flags output. I am not connected to the DragonOS project. I suppose they watch tags and make a new release when a new tags pops out (here 6.1.0) therefore they just pick up the `debian/rules` of the moment.

Cmake files have been extensively changed in 6.1.0 to meet a more modern and systematic approach kindly contributed by @kasper93 However it turns out that architecture requirements are more stringent for the Debian build which includes SSE 4.2 level extensions at least. This will give reasonable performance improvements on most x86_64 machines used at the moment that are all Nehalem and later architectures. Unfortunately this drops older hardware.

You still have the option to compile from source. Appropriate extensions according to your CPU architecture should be detected.

loewal, 2020-11-23:
Ok, thanks!
I will do that.

Grtz.

kasper93, 2020-11-23:
> Cmake files have been extensively changed in 6.1.0 to meet a more modern and systematic approach kindly contributed by @kasper93 However it turns out that architecture requirements are more [...]

I picked `nehalem`, as it is close to previous `FORCE_SSE41`.  Please note that flags
> flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts rep_good nopl cpuid [...]

doesn't include SSE4.1 either which was "enabled" before my changes,. I guess now it is truly enabled.

As for minimal requirement we could switch to `core2`. I don't there would be much users below that  (which would be P4 and previews).

f4exb, 2020-11-23:
> I guess now it is truly enabled.

Yes I am afraid this was not quite working before
> As for minimal requirement we could switch to core2

Well my original intention was to have SSE 4.2 as a minimum so `nehalem` is fine.

f4exb, 2020-11-24:
Reminder in the Wiki [Hardware requirements](https://github.com/f4exb/sdrangel/wiki/Hardware-requirements)

closed 2020-11-24 by f4exb