f4exb/sdrangel #722: J'ai utilisé le DSD et je trouve qu'il manque un petit peu plus de le chose .

gaspar1987, 2020-11-24:
Bonjour au développeur web je trouve que votre  le SDR que vous il fonctionne super bien j'ai pu l'avoir en mode démo grâce à votre application en ligne j'ai pu obtenir des radios que je n'avais pas dans le spectre radio et certains numérique que je n'ai pas pu entendre aussi ça fonctionne super bien je suis content ce mais il manque certaines applications comme certains démodulateur qu'il faudrait mettre en plus comme du Tetra du  PDW  du   certains décodage qui sont dans certaines versions qui peuvent marcher avec des SDR que vous avez écrit  c'est pour cela que je voudrais que vous faire faisait plusieurs démodulateur qui fasse plaisir à l'écoute de la radio qui marche dans le secteur le secteur de la région parisienne parce que je peux entendre beaucoup de choses mais c'est restreint avec le HF qui est  qui est restreint message c'est les messages qui ne passe pas comme le bébé c'est pour cela qu'il faut mettre un amplificateur comme dans le SDR sharp voilà merci à bientôt je vous remercie pour les nouvelles version si vous créez ça ça peut faire plaisir quand on utilise des SDR

f4exb, 2020-11-24:
Thanks for the praise in French as I get it but here we prefer to express in English for better international compatibility, However I do not get fully what you would desire:
  - Tetra: well there is already an open issue
  - PDW a.k.a. paging OK got it. Maybe it can be grouped with other similar types
  - baby? (bébé) babyphone? what...?
  - which amplifier in SDR#? (amplificateur comme dans le SDR sharp )
  -

f4exb, 2020-11-24:
Tetra; https://github.com/f4exb/sdrangel/issues/108

closed 2020-12-13 by f4exb