f4exb/sdrangel #727: Target "demodvor" links to target "Qt5::Positioning" but the target was not   found.

roelandjansen, 2020-11-27:
not sure if this is a config error on the buildsevre but the latest commits:

commit 445b7c57e21f3aba8a5913fa2a16cbbc23404daa (HEAD -> master, tag: v6.2.0, origin/master, origin/HEAD)
Author: f4exb <f4exb06@gmail.com>
Date:   Thu Nov 26 07:12:01 2020 +0100

-- Configuring done
CMake Error at plugins/channelrx/demodvor/CMakeLists.txt:54 (add_library):
  Target "demodvor" links to target "Qt5::Positioning" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

-- Generating done

f4exb, 2020-11-27:
AFAIK it builds: https://ci.appveyor.com/project/f4exb/sdrangel/builds/36519812 It also builds from source so you have to give more details about how you get this error.

f4exb, 2020-11-27:
yes `add_subdirectory(demodvor)` should be moved inside the 'if" here: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/CMakeLists.txt#L4

commit 80829e14 (2020-11-27): VOR demod: compile only if prerequisites are available. Fixes #727
closed 2020-11-27 by f4exb