f4exb/sdrangel #728: Random crashes

alxchk, 2020-12-05:
I'm trying to use sdangel built from the git. From time to time it crashes while I'm playing with channel decoders. I will put some here.

```
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
#1  0x00007ffff53f1538 in __GI_abort () at abort.c:79
#2  0x00007ffff5449d57 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff5555b9b "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#3  0x00007ffff54517cc in malloc_printerr (str=str@entry=0x7ffff5557f20 "free(): invalid next size (fast)") at malloc.c:5389
#4  0x00007ffff5452be3 in _int_free (av=0x7ffff5588a00 <main_arena>, p=0x555556eac6c0, have_lock=0) at malloc.c:4281
#5  0x00007fffd744db72 in  () at /usr/lib64/dri/iris_dri.so
#6  0x00007fffd70bfb1a in  () at /usr/lib64/dri/iris_dri.so
#7  0x00007fffd71d2298 in  () at /usr/lib64/dri/iris_dri.so
#8  0x00007fffd71d234f in  () at /usr/lib64/dri/iris_dri.so
#9  0x00007fffd71a12ae in  () at /usr/lib64/dri/iris_dri.so
#10 0x00007fffd71d5829 in  () at /usr/lib64/dri/iris_dri.so
#11 0x00007fffd70d99c9 in  () at /usr/lib64/dri/iris_dri.so
#12 0x00007fffd70467e6 in  () at /usr/lib64/dri/iris_dri.so
#13 0x00007fffd702835e in  () at /usr/lib64/dri/iris_dri.so
#14 0x00007fffd76eff73 in  () at /usr/lib64/dri/iris_dri.so
#15 0x00007ffff018925f in  () at /usr/lib64/libGLX_mesa.so.0
#16 0x00007ffff0173479 in  () at /usr/lib64/libGLX_mesa.so.0
#17 0x00007ffff040af95 in  () at /usr/lib64/qt5/plugins/xcbglintegrations/libqxcb-glx-integration.so
#18 0x00007ffff5fb6a0b in QOpenGLContext::destroy() () at /usr/lib64/libQt5Gui.so.5
#19 0x00007ffff5fb6c87 in QOpenGLContext::~QOpenGLContext() () at /usr/lib64/libQt5Gui.so.5
#20 0x00007ffff5fb6c99 in QOpenGLContext::~QOpenGLContext() () at /usr/lib64/libQt5Gui.so.5
#21 0x00007ffff6ac3887 in QGLContext::reset() () at /usr/lib64/libQt5OpenGL.so.5
#22 0x00007ffff6ac5478 in QGLContext::~QGLContext() () at /usr/lib64/libQt5OpenGL.so.5
#23 0x00007ffff6ac57cd in QGLWidget::~QGLWidget() () at /usr/lib64/libQt5OpenGL.so.5
#24 0x00007ffff7e05429 in TVScreen::~TVScreen() () at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/sdrgui/gui/tvscreen.cpp:54
#25 0x00007ffff5bbbc47 in QObjectPrivate::deleteChildren() () at /usr/lib64/libQt5Core.so.5
#26 0x00007ffff65a6086 in QWidget::~QWidget() () at /usr/lib64/libQt5Widgets.so.5
#27 0x00007ffff65a6279 in QWidget::~QWidget() () at /usr/lib64/libQt5Widgets.so.5
#28 0x00007ffff5bbbc47 in QObjectPrivate::deleteChildren() () at /usr/lib64/libQt5Core.so.5
#29 0x00007ffff65a6086 in QWidget::~QWidget() () at /usr/lib64/libQt5Widgets.so.5
#30 0x00007fffd0068a39 in DSDDemodGUI::~DSDDemodGUI() () at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demoddsd/dsddemodgui.cpp:386
#31 0x00007ffff5bbd417 in QObject::event(QEvent*) () at /usr/lib64/libQt5Core.so.5
#32 0x00007ffff657d55f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#33 0x00007ffff5b9cc88 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib64/libQt5Core.so.5
#34 0x00007ffff5b9ce7a in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib64/libQt5Core.so.5
#35 0x00007ffff5be8a73 in  () at /usr/lib64/libQt5Core.so.5
#36 0x00007ffff4fb8b5b in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#37 0x00007ffff5009648 in  () at /usr/lib64/libglib-2.0.so.0
#38 0x00007ffff4fb73ff in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#39 0x00007ffff5be46cb in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#40 0x00007ffff5b9b3fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#41 0x00007ffff5b9d200 in QCoreApplication::exec() () at /usr/lib64/libQt5Core.so.5
#42 0x0000555555556e08 in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) (argc=1, argv=0x7fffffffd230, logger=0x0) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/app/main.cpp:130
#43 0x0000555555556437 in main(int, char**) (argc=1, argv=0x7fffffffd5a8) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/app/main.cpp:137
```

alxchk, 2020-12-05:
```
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:307
307		movl	%ecx, -4(%rdi,%rdx)
(gdb) bt
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:307
#1  0x00007fff98aa0017 in freedv_data_channel_rx_frame () at /usr/lib64/libcodec2.so.0.9
#2  0x00007fff98aa049a in  () at /usr/lib64/libcodec2.so.0.9
#3  0x00007fff98aa07b4 in fvhff_deframe_bits () at /usr/lib64/libcodec2.so.0.9
#4  0x00007fff98aaa0e8 in freedv_comprx_fsk () at /usr/lib64/libcodec2.so.0.9
#5  0x00007fff98aad6df in freedv_comprx () at /usr/lib64/libcodec2.so.0.9
#6  0x00007fff98aada94 in freedv_floatrx () at /usr/lib64/libcodec2.so.0.9
#7  0x00007fff98aae165 in freedv_rx () at /usr/lib64/libcodec2.so.0.9
#8  0x00007fff91e33eda in FreeDVDemodSink::pushSampleToDV(short) (this=0x555555f4c080, sample=-6051) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:293
#9  0x00007fff91e34115 in FreeDVDemodSink::processOneSample(std::complex<float>&) (this=0x555555f4c080, ci=...) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:271
#10 0x00007fff91e34931 in FreeDVDemodSink::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555555f4c080, begin=<error reading variable: Cannot access memory at address 0x54aaaa57>, end={m_real = 900, m_imag = 7162})
    at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:206
#11 0x00007ffff7558709 in DownChannelizer::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555555f25980, begin=<error reading variable: Cannot access memory at address 0x54aaaa57>, end={m_real = -8044, m_imag = 8700}) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/sdrbase/dsp/downchannelizer.cpp:85
#12 0x00007fff91e31032 in FreeDVDemodBaseband::handleData() (this=0x555555f4c010) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodbaseband.cpp:83
#13 0x00007ffff5bbd685 in QObject::event(QEvent*) () at /usr/lib64/libQt5Core.so.5
#14 0x00007ffff657d55f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#15 0x00007ffff5b9cc88 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib64/libQt5Core.so.5
#16 0x00007ffff5b9ce7a in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib64/libQt5Core.so.5
#17 0x00007ffff5be8a73 in  () at /usr/lib64/libQt5Core.so.5
#18 0x00007ffff4fb8b5b in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#19 0x00007ffff5009648 in  () at /usr/lib64/libglib-2.0.so.0
#20 0x00007ffff4fb73ff in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#21 0x00007ffff5be46cb in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#22 0x00007ffff5b9b3fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#23 0x00007ffff5a202b7 in QThread::exec() () at /usr/lib64/libQt5Core.so.5
#24 0x00007ffff5a21892 in  () at /usr/lib64/libQt5Core.so.5
#25 0x00007ffff50faf9e in start_thread (arg=0x7fff7f7fe640) at pthread_create.c:463
#26 0x00007ffff54ca0cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

alxchk, 2020-12-06:
```
Thread 27 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff7f053640 (LWP 74648)]
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:300
300		movq	%rcx, -8(%rdi,%rdx)
(gdb) bt
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:300
#1  0x00007fffb0a4b017 in freedv_data_channel_rx_frame () at /usr/lib64/libcodec2.so.0.9
#2  0x00007fffb0a4b49a in  () at /usr/lib64/libcodec2.so.0.9
#3  0x00007fffb0a4b6ac in fvhff_deframe_bits () at /usr/lib64/libcodec2.so.0.9
#4  0x00007fffb0a550e8 in freedv_comprx_fsk () at /usr/lib64/libcodec2.so.0.9
#5  0x00007fffb0a586df in freedv_comprx () at /usr/lib64/libcodec2.so.0.9
#6  0x00007fffb0a58a94 in freedv_floatrx () at /usr/lib64/libcodec2.so.0.9
#7  0x00007fffb0a59165 in freedv_rx () at /usr/lib64/libcodec2.so.0.9
#8  0x00007fff91e2deda in FreeDVDemodSink::pushSampleToDV(short) (this=0x555555f41b90, sample=1561) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:293
#9  0x00007fff91e2e115 in FreeDVDemodSink::processOneSample(std::complex<float>&) (this=0x555555f41b90, ci=...) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:271
#10 0x00007fff91e2e931 in FreeDVDemodSink::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555555f41b90, begin=<error reading variable: Cannot access memory at address 0xe59fbbbeaeae>, end={m_real = 0, m_imag = 0})
    at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:206
#11 0x00007ffff7558709 in DownChannelizer::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555555f2d580, begin=<error reading variable: Cannot access memory at address 0xe59fbbbeaeae>, end={m_real = 14872, m_imag = -16932})
    at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/sdrbase/dsp/downchannelizer.cpp:85
#12 0x00007fff91e2b032 in FreeDVDemodBaseband::handleData() (this=0x555555f41b20) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodbaseband.cpp:83
#13 0x00007ffff5bbd685 in QObject::event(QEvent*) () at /usr/lib64/libQt5Core.so.5
#14 0x00007ffff657d55f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#15 0x00007ffff5b9cc88 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib64/libQt5Core.so.5
#16 0x00007ffff5b9ce7a in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib64/libQt5Core.so.5
#17 0x00007ffff5be8a73 in  () at /usr/lib64/libQt5Core.so.5
#18 0x00007ffff4fb8b5b in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#19 0x00007ffff5009648 in  () at /usr/lib64/libglib-2.0.so.0
#20 0x00007ffff4fb73ff in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#21 0x00007ffff5be46cb in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#22 0x00007ffff5b9b3fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#23 0x00007ffff5a202b7 in QThread::exec() () at /usr/lib64/libQt5Core.so.5
#24 0x00007ffff5a21892 in  () at /usr/lib64/libQt5Core.so.5
#25 0x00007ffff50faf9e in start_thread (arg=0x7fff7f053640) at pthread_create.c:463
#26 0x00007ffff54ca0cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

f4exb, 2020-12-08:
There is not enough information to address the issue properly:

- What is the context: OS distribution, Qt version ?
- How is the software built ? Is it following [these instructions](https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux) thoroughly ?
- The first core relates to issue in Qt OpenGL when closing a plugin? Which one is it? This does not seem related to an issue in the code but rather some Qt compatibility issue or a bug with Qt (5.12 is known to have bugs that have been corrected in 5.14).
- The second and third core relates to an issue with the freedv library so again does not seem to be directly related to an issue with the code. How did you obtain the freedv library?

alxchk, 2020-12-08:
- Gentoo, deps from git, QT 5.15.1
- As I don't have any other softare which uses these libraries, I made ebuilds and install them to the system. The rest should be close enough to the instructions (at least it's not really obvious why it should go wrong because of the build).
- Codec2 was v0.9.2, i.e. not that revision which is specified in build instructions. So I rebuilt it with that rev (76a20416d715ee06f8b36a9953506876689a3bd2; also with debug). Can reproduce:

```
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:300
300		movq	%rcx, -8(%rdi,%rdx)
(gdb) bt
#0  __memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:300
#1  0x00007fff98a9bef7 in memcpy (__len=14, __src=<optimized out>, __dest=<optimized out>, __dest=<optimized out>, __src=<optimized out>, __len=<optimized out>) at /usr/include/bits/string_fortified.h:34
#2  freedv_data_channel_rx_frame (fdc=0x7fff7401c290, data=<optimized out>, data@entry=0x7fff7e0fb490 "\216WQb\025\265", size=size@entry=6, from_bit=from_bit@entry=0, bcast_bit=<optimized out>, crc_bit=<optimized out>, end_bits=14)
    at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_data_channel.c:196
#3  0x00007fff98a9c37a in fvhff_extract_frame_data (def=def@entry=0x7fff74008180, bits=bits@entry=0x7fffec012610 "\001")
    at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_vhf_framing.c:727
#4  0x00007fff98a9c694 in fvhff_extract_frame (pt=<optimized out>, vc_out=0x7fff7e0fb5b3 "?\200\232l>", proto_out=0x7fff7e0fb5b5 "\232l>", codec2_out=0x7fff74008160 "\220", bits=0x7fffec012610 "\001", def=0x7fff74008180)
    at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_vhf_framing.c:739
#5  fvhff_deframe_bits (def=0x7fff74008180, codec2_out=0x7fff74008160 "\220", proto_out=proto_out@entry=0x7fff7e0fb5b5 "\232l>", vc_out=vc_out@entry=0x7fff7e0fb5b3 "?\200\232l>", bits_in=<optimized out>)
    at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_vhf_framing.c:848
#6  0x00007fff98aa5f68 in freedv_comprx_fsk (f=f@entry=0x7fff740115f0, demod_in=demod_in@entry=0x7fff7e0fb6a0, valid=valid@entry=0x7fff7e0fb634)
    at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_api.c:1625
#7  0x00007fff98aa94bf in freedv_comprx (f=f@entry=0x7fff740115f0, speech_out=speech_out@entry=0x7fff7400fe40, demod_in=0x7fff7e0fb6a0)
    at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_api.c:2466
#8  0x00007fff98aa9874 in freedv_floatrx (f=f@entry=0x7fff740115f0, speech_out=speech_out@entry=0x7fff7400fe40, demod_in=demod_in@entry=0x7fff7e0fcbb0)
    at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_api.c:1671
#9  0x00007fff98aa9f45 in freedv_rx (f=0x7fff740115f0, speech_out=0x7fff7400fe40, demod_in=0x7fff7401dc60) at /usr/src/debug/media-libs/codec2-0.8.9999-r1/codec2-76a20416d715ee06f8b36a9953506876689a3bd2/src/freedv_api.c:1570
#10 0x00007fff91c7deda in FreeDVDemodSink::pushSampleToDV(short) (this=0x555556ba84f0, sample=-1288) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:293
#11 0x00007fff91c7e115 in FreeDVDemodSink::processOneSample(std::complex<float>&) (this=0x555556ba84f0, ci=...) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:271
#12 0x00007fff91c7e931 in FreeDVDemodSink::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555556ba84f0, begin=<error reading variable: Cannot access memory at address 0xb5156251578e>, end={m_real = 148, m_imag = -242})
    at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodsink.cpp:206
#13 0x00007ffff7554c49 in DownChannelizer::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x5555567275f0, begin=<error reading variable: Cannot access memory at address 0xb5156251578e>, end={m_real = -9008, m_imag = 564})
    at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/sdrbase/dsp/downchannelizer.cpp:85
#14 0x00007fff91c7b032 in FreeDVDemodBaseband::handleData() (this=0x555556ba8480) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodfreedv/freedvdemodbaseband.cpp:83
#15 0x00007ffff5bac685 in QObject::event(QEvent*) () at /usr/lib64/libQt5Core.so.5
#16 0x00007ffff656c55f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#17 0x00007ffff5b8bc88 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib64/libQt5Core.so.5
#18 0x00007ffff5b8be7a in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib64/libQt5Core.so.5
#19 0x00007ffff5bd7a73 in  () at /usr/lib64/libQt5Core.so.5
#20 0x00007ffff4fa7b5b in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#21 0x00007ffff4ff8648 in  () at /usr/lib64/libglib-2.0.so.0
[... cut ...]
3. Using mouse wheel quickly switch FreeDV mode in all directions :) Usually it crashes after 1-5 seconds of switches. If not - I'm moving frequency randomly..

> The first core relates to issue in Qt OpenGL when closing a plugin. Which one is it? This does not seem related to an issue in the code but rather some Qt compatibility issue or a bug with Qt (5.12 is known to have bugs in the OpenGL support that have been corrected in 5.14).

I'm not sure I understand the question correctly. I do have Intel HD620 and mesa for opengl

f4exb, 2020-12-08:
The issue with FreeDV demod is apparently caused by this line of code:
https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodfreedv/freedvdemodsink.cpp#L462

It turns out that if data blocks are not used then this function should not be called at all. I suppose it is the same with the two lines above. When removing or commenting these 3 lines the issue seems to disappear.

f4exb, 2020-12-09:
For the QOpenGL thing I initially missed that `DSDDemodGUI::~DSDDemodGUI()` was up in the stack trace so this should come from closing the DSDDemod window. However I cannot reproduce it and have no clue about what it may relate in the code (unlike the FreeDV thing) so I am afraid I will not be able to address it properly.

alxchk, 2020-12-09:
I will try to catch it some how once again. For now I have new one with ATV. This once crashes immediately after adding. 
I.e. it looks like this: I point to some random freq with white noise, activate receiver, add channel -> crash.

```
Thread 31 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff7f4f6640 (LWP 933635)]
ATVDemodSink::demod (this=0x555556513bd0, c=...) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/sdrgui/gui/tvscreenanalog.h:97
97			if ((column < m_width - 2) && (column >= -2))

(gdb) bt
#0  ATVDemodSink::demod(std::complex<float>&) (this=0x555556513bd0, c=...) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/sdrgui/gui/tvscreenanalog.h:97
#1  0x00007fffd01e09cf in ATVDemodSink::feed(__gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&, __gnu_cxx::__normal_iterator<Sample const*, std::vector<Sample, std::allocator<Sample> > > const&) (this=0x555556513bd0, begin=<error reading variable: Cannot access memory at address 0xc734d400466b9000>, end={m_real = -137252, m_imag = -37880})
    at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodatv/atvdemodsink.cpp:95
#2  0x00007fffd01de97f in ATVDemodBaseband::handleData() (this=0x555556513b60) at /usr/src/debug/net-wireless/sdrangel-9999/sdrangel-9999/plugins/channelrx/demodatv/atvdemodbaseband.cpp:97
#3  0x00007ffff5bab685 in QObject::event(QEvent*) () at /usr/lib64/libQt5Core.so.5
#4  0x00007ffff656b55f in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib64/libQt5Widgets.so.5
#5  0x00007ffff5b8ac88 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib64/libQt5Core.so.5
#6  0x00007ffff5b8ae7a in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /usr/lib64/libQt5Core.so.5
#7  0x00007ffff5bd6a73 in  () at /usr/lib64/libQt5Core.so.5
#8  0x00007ffff4fa6b5b in g_main_context_dispatch () at /usr/lib64/libglib-2.0.so.0
#9  0x00007ffff4ff7648 in  () at /usr/lib64/libglib-2.0.so.0
#10 0x00007ffff4fa53ff in g_main_context_iteration () at /usr/lib64/libglib-2.0.so.0
#11 0x00007ffff5bd26cb in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#12 0x00007ffff5b893fa in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib64/libQt5Core.so.5
#13 0x00007ffff5a0e2b7 in QThread::exec() () at /usr/lib64/libQt5Core.so.5
#14 0x00007ffff5a0f892 in  () at /usr/lib64/libQt5Core.so.5
#15 0x00007ffff50e8f9e in start_thread (arg=0x7fff7f4f6640) at pthread_create.c:463
#16 0x00007ffff54b80cf in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

```
(gdb) info locals
sampleNormI = 0.309663773
sampleNormQ = 0.5
sampleNorm = <optimized out>
sample = 0
sampleVideo = <optimized out>
magSq = <optimized out>
(gdb) print *this
$9 = {m_channelSampleRate = 1700000, m_channelFrequencyOffset = 0, m_samplesPerLine = 108, m_samplesPerLineFrac = 0.800003052, m_settings = {m_inputFrequencyOffset = 0, m_bfoFrequency = 0, m_atvModulation = 0, m_fmDeviation = 0.5, 
    m_amScalingFactor = 100, m_amOffsetFactor = 0, m_fftFiltering = false, m_fftOppBandwidth = 0, m_fftBandwidth = 6000, m_nbLines = 625, m_fps = 25, m_atvStd = 0, m_hSync = false, m_vSync = false, m_invertVideo = false, 
    m_halfFrames = false, m_levelSynchroTop = 0.150000006, m_levelBlack = 0.300000012, m_rgbColor = 4294967295, m_title = {d = 0x5555560f7fc0}, m_udpAddress = {d = 0x555556f35790}, m_udpPort = 9999, m_channelMarker = 0x0, 
    m_streamIndex = 0, m_rfSliderDivisor = 0}, m_videoTabIndex = 0, m_scopeSink = 0x555556594680, Python Exception <class 'RuntimeError'> Type is not a template.: 
m_scopeSampleBuffer = {<No data fields>}, m_registeredTVScreen = 0x555556509eb0, Python Exception <class 'gdb.error'> There is no member named _M_ptr.: 
m_tvScreenBuffer = {<No data fields>}, 
  m_numberSamplesPerHTop = 7, m_numberOfBlackLines = 49, m_firstVisibleLine = 23, m_fieldDetectStartPos = 3, m_fieldDetectEndPos = 46, m_vSyncDetectStartPos = 58, m_vSyncDetectEndPos = 100, m_vSyncDetectThreshold = 21, 
  m_fieldDetectThreshold1 = 31, m_fieldDetectThreshold2 = 10, m_numberOfVSyncLines = 3, m_numberSamplesPerLineSignals = 20, m_numberSamplesPerHSync = 17, m_numberSamplesHSyncCrop = 9, m_interleaved = true, m_fieldIndex = 0, 
  m_synchroSamples = 0, m_fieldDetectSampleCount = 0, m_vSyncDetectSampleCount = 0, m_effMin = 20, m_effMax = -20, m_ampMin = -1, m_ampMax = 1, m_ampDelta = 2, m_fltBufferI = {0.309663773, 0.761551201, 0, 0, 0, 0}, m_fltBufferQ = {
    -0.950846076, 0.648104727, 0, 0, 0, 0}, m_amSampleIndex = 0, m_sampleOffset = 0, m_sampleOffsetFrac = 0, m_sampleOffsetDetected = 0, m_lineIndex = 0, m_hSyncShift = 0, m_hSyncErrorCount = 0, prevSample = 4.59163468e-41, 
  m_avgColIndex = -164872112, Python Exception <class 'RuntimeError'> Type is not a template.: 
m_sampleBuffer = {<No data fields>}, m_sampleRangeCorrection = 364.285706, m_magSqAverage = {m_samples = {2370235136, 13722871808, 13370569728, 6394689024, 110518059008, 6083213312, 49639305216, 
      25935546368, 5136697856, 48358498304, 14906135552, 49755537408, 3250163456, 5070340608, 124520488960, 407605504, 128723042304, 1826190208, 21634365440, 5280239616, 15815392256, 52128419840, 3085078272, 6324890624, 10614047744, 
[... cut ...]
    m_fmDeviation = 0.5, m_amScalingFactor = 100, m_amOffsetFactor = 0, m_fftFiltering = false, m_fftOppBandwidth = 0, m_fftBandwidth = 6000, m_nbLines = 625, m_fps = 25, m_atvStd = 0, m_hSync = false, m_vSync = false, 
    m_invertVideo = false, m_halfFrames = false, m_levelSynchroTop = 0.150000006, m_levelBlack = 0.300000012, m_rgbColor = 4294967295, m_title = {d = 0x5555560f7fc0}, m_udpAddress = {d = 0x555556f35790}, m_udpPort = 9999, 
    m_channelMarker = 0x0, m_streamIndex = 0, m_rfSliderDivisor = 0}, m_running = true, m_mutex = {<No data fields>}}
(gdb) 
```

f4exb, 2020-12-10:
Issue is because here: https://github.com/f4exb/sdrangel/blob/04dbb028aa541f2e8c76d1d9e7ae85be76a649db/plugins/channelrx/demodatv/atvdemodsink.h#L48
`m_tvScreenBuffer` is not initialized. This should become:
```c++
    void setTVScreen(TVScreenAnalog *tvScreen) //!< set by the GUI
    {
        m_registeredTVScreen = tvScreen;
        m_tvScreenBuffer = m_registeredTVScreen->getBackBuffer();
    }
```

f4exb, 2020-12-10:
Also the usage of `std::shared_ptr` for QOpenGL things appear to cause these hard to debug issues at delete time:
https://github.com/f4exb/sdrangel/blob/04dbb028aa541f2e8c76d1d9e7ae85be76a649db/sdrgui/gui/tvscreenanalog.h#L135

I get segfaults systematically when closing the `ATVDemod` window while the device is running. When using classical `new` -> `delete` sequence this does not seem to occur.

It may also explain the issue when closing the `DSDDemod` window.

f4exb, 2020-12-12:
Fixes except TVScreen deletion issue have been implemented in v6.3.1

There could be something fishy with TVScreen but since this is unreproducible it is not possible to investigate it properly.

alxchk, 2020-12-12:
Thanks. I already trying with latest master. Don't have these crashes for now

f4exb, 2020-12-12:
I have an idea for TVScreen but I am unable to test it as it seems to work in my environment anyway. In TVScreen the cleanup() method is connected to QOpenGLContext deletion this way:
https://github.com/f4exb/sdrangel/blob/38a06c4f6949c46b327add392920dbd4dd9a3812/sdrgui/gui/tvscreen.cpp#L151
And cleanup() is also called in the destructor:
https://github.com/f4exb/sdrangel/blob/38a06c4f6949c46b327add392920dbd4dd9a3812/sdrgui/gui/tvscreen.cpp#L53
This could be one time too much I think it may not be called in the destructor. This is not the case in TVScreenAnalog:
https://github.com/f4exb/sdrangel/blob/38a06c4f6949c46b327add392920dbd4dd9a3812/sdrgui/gui/tvscreenanalog.cpp

f4exb, 2020-12-12:
This latest fix was implemented in v6.3.2

f4exb, 2020-12-20:
Regression introduced by https://github.com/f4exb/sdrangel/commit/a08e18b505c2f1baa1dcac76dae7991d304a878b

f4exb, 2020-12-21:
Fixed in v6.4.0

closed 2020-12-21 by f4exb